project-chip/connectedhomeip #26034: [Platform] Standardize python tests to be able to run them in Test Harness UI

ccruzagralopes, 2023-04-10:
### Reproduction steps / Feature

The Test Harness team received a request to support running python tests in the UI. In order to do that, we need some basic test case information.

The information below must be present in the python test files:
- [ ] Metadata (public_id, version, title and description)
    - Needed to present the test for selection in TH UI
- [ ] Default test parameters (e.g.: nodeId, cluster, endpoint, timeout...)
    - Some of them are hardcoded in the python tests
    - So for these tests to follow the same execution pattern as the current TH test cases, they would need to allow these parameters to be overridden
- [ ] PICS (top level)
    - Needed for TH to filter the applicable test cases
- [ ] Test step descriptions
    - They are present in the tests but they don’t follow a specific pattern
    - In order to properly indicate the current executing step, TH needs the step log to follow a specific pattern: “***** Test Step <step number> : <step description>”

Some other things to consider:
- [ ] It seems like these python tests are not handling PICS in step level, which could cause failures.
    - If a step is not applicable because of PICS, TH expects a log entry pattern to indicate that the step skipped: “**** Skipping: <test step>”
- [ ] For each prompt step, we need some indication that that step needs a prompt
    - Needed by the test wrapper
- [ ] If a step requires a user prompt, it must indicate with a log that contains "USER_PROMPT"
    - Parsed during execution to present the log

### Platform

python

### Type

Manually tested with SDK

### Anything else?

Link to TH issue: https://github.com/CHIP-Specifications/chip-certification-tool/issues/52

cecille, 2023-05-18:
General question - it sounds like the plan is to have the TH interact with the scripts via log scraping to present a summary of the steps. Is this correct? Is there maybe an easier way to do this? Visually linting python scripts to ensure that the conform to the required log details isn't robust. Is there maybe a programatic way we can get this information passed?

what's the metadata and where is it stored?

ccruzagralopes, 2023-05-19:
Hi Cecile,
We agree that log scraping is not the best way to do this, but we thought it would be the solution with less changes. The ideal approach would be do something like the runner that's being developed by @krypton36 so we have the same interface for python and YAML tests. Does that make sense to you?

> what's the metadata and where is it stored?

The metadata is: public_id, version, title and description. Like the YAML scripts provides us. They are need so that we can list the tests in TH UI.

cecille, 2023-05-30:
Where is the interface for the runner documented?

cecille, 2023-05-31:
Does the name and number actually get logged when running the test? Where is this scraped from in the TH?

ccruzagralopes, 2023-05-31:
> Where is the interface for the runner documented?

I'm not sure if it it. @krypton36 do you know?

> Does the name and number actually get logged when running the test? Where is this scraped from in the TH?

No, they are not logged. We were thinking of parsing the test file to get this information. We need this information to list the test cases in the UI for the user to select the ones they'd like to run. So we need access to it before the test is running.

krypton36, 2023-05-31:
Hello @cecille, we are integrating [this new runner](https://github.com/project-chip/connectedhomeip/blob/master/scripts/py_matter_yamltests/matter_yamltests/websocket_runner.py). I don't believe there is documentation on it, but it appears that there is something sort of using the new runner we are trying to integrate [here](https://github.com/project-chip/connectedhomeip/blob/6d5111ba78f6d61a8c9b18469b65f3af00aa5be4/scripts/tests/chiptest/yamltest_with_chip_repl_tester.py#L58) with the Python CHIP controller implementation. Ideally, the test harness will need the same hooks as an option and being called as the [base runner implementation](https://github.com/project-chip/connectedhomeip/blob/master/scripts/py_matter_yamltests/matter_yamltests/runner.py) does defined for TestRunnerHooks, which is defined in [here](https://github.com/project-chip/connectedhomeip/blob/master/scripts/py_matter_yamltests/matter_yamltests/hooks.py#L78). So for example these [lines](https://github.com/project-chip/connectedhomeip/blob/6d5111ba78f6d61a8c9b18469b65f3af00aa5be4/src/python_testing/TC_ACE_1_3.py#LL97C1-L98C55) would look some like this:
```python
self.print_step(3, "TH1 reads EP0 descriptor - expect SUCCESS")
if not request.is_pics_enabled:
    test_harness_hook.step_skipped(request.label, request.pics)
else:
    test_harness_hook.step_start(request)
    await self.read_descriptor_expect_success(TH1)
    if success:
        test_harness_hook.step_success(logger, logs, duration, request)
    else:
        test_harness_hook.step_failure(logger, logs, duration, request, received)
```
Finally, the test harness needs a YAML matching the test case information with the test steps to stay in sync and present the test in UI and @ccruzagralopes mentioned previously.

cecille, 2023-06-02:
Using a separate YAML with just the test steps listed seems like a recipe for having the yaml and the python fall out of sync. Let me think on a better way to do this.

The python scripts right now use a main function that relies on having a single instance of the MatterBaseTest. I'm not sure how nicely this will play with the TH. They would also need to have the hook passed in somehow. 

I have a draft of the changes that will make the requested prints work. I will poke at integrating the hook object today, just by instantiating it from within the script. But we will have to coordinate on how the TH is planning to load the scripts etc. I would prefer if the test step information comes from within the python script so we don't need to manually keep two separate pieces of code in sync.

krypton36, 2023-06-06:
Hello @cecille the test harness can accept the test steps as a python script. [Here](https://github.com/CHIP-Specifications/chip-certification-tool-backend/blob/develop/test_collections/sample_tests/sample_suite_1/tcss1001/tcss1001.py) is an example of a defined test with steps. The issue I see here is that since you are contributing to open source, you will not have access to the device [TestStep](https://github.com/CHIP-Specifications/chip-certification-tool-backend/blob/e5fc8ba56882bb9f778b459f428eac5bba2d4dc7/app/test_engine/models/test_step.py#L9) object defined in the TestHarness. One way can think of is to implement the test in such a way where it will import and use these classes if available, but this doesn't seem too feasible. Another option is to have the tests directly in the Test Harness repo, but not sure who to reach out for that.

raju-apple, 2023-06-14:
@krypton36 now that you have the phase 2 changes ready on a branch can you share it with @cecille so that she can try it out and it could kick start getting these python tests into the TH UI as well.