project-chip/connectedhomeip #26041: [Platform] [Linux] A deadlock occurs in the commission of the device

yzm157, 2023-04-11:
### Reproduction steps

1. Turn off all Matter devices
2. Use Python for device commission

```
# ..... Init devCtrl
devCtrl.CommissionWiFi(3840, 20202021, 1, "wifi-ssid", "wifi-password")    # deadlock
```

Deadlock caused by not handling bluetooth timeout.

At src/platform/Linux/BLEManagerImpl.cpp
```
void BLEManagerImpl::OnScanError(CHIP_ERROR err)
{
    ChipLogError(Ble, "BLE scan error: %" CHIP_ERROR_FORMAT, err.Format());
}
```
The failure reason is not called back to BleLayer

### Platform

python, raspi

### Type

Platform validated

tianfeng-yang, 2023-05-06:
at src/controller/CHIPDeviceController.cpp

tianfeng-yang, 2023-05-06:
@yzm157 @bzbarsky-apple 

at src/controller/CHIPDeviceController.cpp

```
1. DeviceCommissioner::PairDevice(NodeId remoteDeviceId, RendezvousParameters & rendezvousParams, CommissioningParameters & commissioningParams)
2. DeviceCommissioner::EstablishPASEConnection(NodeId remoteDeviceId, RendezvousParameters & params)
3. DeviceCommissioner::FindCommissioneeDevice(const Transport::PeerAddress & peerAddress)
```

blocked in step3, FindCommissioneeDevice not set timeout. There is an infinite loop here

PR #26442 (2023-05-10): Fix the problem of hang after BLE discovery fails during commission
Fixes #26041

On the Linux platform, the BLE discovery failure was not handled during the Commission.
closed 2023-05-10 by andy31415