project-chip/connectedhomeip #26043: [BUG] [NXP] [lighting-app] build error

mmt-10, 2023-04-11:
### Reproduction steps

I followed [building](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0#building).

And, I change `out/debug` to `./out/debug/`.

1. download [K32W061DK6 SDK 2.6.10](https://cache.nxp.com/lgfiles/bsps/SDK_2_6_10_K32W061DK6.zip)
2. mkdir ~/matter/tools/NXP/k32w
3. cd ~/matter/tools/NXP/k32w
4. mv ~/Downloads/SDK_2_6_10_K32W061DK6.zip .
5. unzip SDK_2_6_10_K32W061DK6.zip
6. export NXP_K32W0_SDK_ROOT=$PWD
7. cd ${MATTER-SDK}
8. source scripts/activate.sh
9. cd examples/lighting-app/nxp/k32w/k32w0
10. cd gn gen ./out/debug --args="k32w0_sdk_root=\"${NXP_K32W0_SDK_ROOT}\" chip_with_OM15082=1 chip_with_ot_cli=0 is_debug=false chip_crypto=\"platform\" chip_with_se05x=0 chip_pw_tokenizer_logging=true"
11. ninja -C ./out/debug

then, error occurs.
```bash
ninja: Entering directory `./out/debug/'
[1059/1074] ld ./chip-k32w0x-light-example
FAILED: chip-k32w0x-light-example chip-k32w0x-light-example.map 
...
collect2: error: ld returned 1 exit status
[1060/1074] ACTION //third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup:pigweed_build..._virtualenv(//third_party/connectedhomeip/third_party/pigweed/repo/pw_build/python_toolchain:python)
ninja: build stopped: subcommand failed.
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 4088a77f557e8571a39338fad51a1d8eb0131d79 (HEAD, tag: v1.0.0.2)

### Platform

k32w

### Anything else?

### Environment
host OS : ubuntu 22.04

bzbarsky-apple, 2023-04-11:
> then, error occurs.

@mmt-10 In that log bit, is there more text where "..." is? If so, could you attach (not paste) the full log without redactions?

mmt-10, 2023-04-11:
@bzbarsky-apple 
I add Full Log to top of this thread.

bzbarsky-apple, 2023-04-12:
Ah, ok.  So the issue is the `chip_crypto="platform"` bits in the build command above:

```
/home/mmt-10/matter/tools/NXP/k32w/middleware/wireless/ble_controller/config/controller_config.c:357: undefined reference to `ECDH_P256_GenerateKeys'
...
/home/mmt-10/matter/matter-v1.0.0.2/examples/lighting-app/nxp/k32w/k32w0/out/debug/../../third_party/connectedhomeip/src/app/util/attribute-storage.cpp:132: undefined reference to `chip::Crypto::DRBG_get_bytes(unsigned char*, unsigned int)'
...
```

etc.

@doru91 Could you please take a look at the NXP platform crypto bits here?

marius-alex-tache, 2023-04-13:
Hi. You are trying to build an example from an old commit (v1.0.0.2 tag), but with latest K32W0 SDK (2.6.10).

What we would recommend is either one of these:
- use the top of the tree, which contains latest K32W0 changes.
- use the following release https://github.com/NXP/matter/releases/tag/v1.0.0.2.1, which contains latest K32W0 changes on top of v1.0.0.2 tag.

mmt-10, 2023-04-14:
Hi,
Thank you for your reply!
I tried `https://github.com/NXP/matter` as below.
I could build without error.

```bash
git clone --recurse-submodules -b v1.0.0.2.1 https://github.com/NXP/matter.git matter-nxp-v1.0.0.2.1
```

I followed developing step from below URL.
[https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0)
It does not specify `https://github.com/NXP/matter.git`, so I confused. 

BTW,
I got error on `git clone` as below.

> fatal: Failed to recurse into submodule path 'third_party/openthread/ot-nxp'

Is it no problem?

And, `DK6Programmer.exe` is only for windows?
How do I flash the bin file to the K32W under Ubuntu Linux?

mmt-10, 2023-04-14:
Hi, I have additional confirmation.

I could flash bin file from DK6Programmer.exe on windows PC.

I tried below command.
I pressed SW2, SW4 and RESET, However, I could not get log / message from DK006 board.

```bash
python3 ../../../../../examples/platform/nxp/k32w/k32w0/scripts/detokenizer.py serial -i /dev/ttyACM0 -d out/debug/chip-k32w0x-light-example-database.bin -o device.txt 
```

How do I get log from DK006 board?
Which USB connect to PC?
And, please disclose jumper settings.

[image]

marius-alex-tache, 2023-04-17:
Hi and sorry for the late reply (whole team is in vacation). OTA is enabled by default, so you also have to flash a SSBL (in case you haven't done it already). You can find the default version in the SDK, but you could also compile it - please see [build SSBL](https://github.com/NXP/matter/tree/v1.0.0.2-tag-nxp/examples/lighting-app/nxp/k32w/k32w0#writing-the-ssbl).

For serial logs on Windows, please use USB B, with jumpers JP4/JP7 in the middle position (UART0-FTDI). Please see the following guide, which is referenced by the lighting-app README: https://github.com/openthread/ot-nxp/tree/main/src/k32w0/k32w061#flash-binaries

mmt-10, 2023-04-19:
Thank you for your reply.

I tried it, but, K32W serial does not output message.
Something wrong, I'm under checking.

If you have any suggestion, please let me know.

[log.txt](https://github.com/project-chip/connectedhomeip/files/11278269/log.txt)

marius-alex-tache, 2023-04-20:
Hi. Your steps seem correct. Could you also try to flash the following SSBL: `boards/k32w061dk6/wireless_examples/framework/ssbl/binary/ssbl_ext_flash_pdm_support.bin`?

mmt-10, 2023-04-21:
Hi,
Thank you for your reply.
I changed `k32w061dk6_ssbl.bin` to ` ssbl_ext_flash_pdm_support.bin`.
Then, LED2 blink.
And, k32w serial message is working.

```
set COMPORT=COM33

C:\NXP\DK6Programmer\DK6Programmer.exe -V 5 -P 1000000 -s %COMPORT% -e Flash

REM C:\NXP\DK6Programmer\DK6Programmer.exe -V2 -s %COMPORT% -P 1000000 -Y -p FLASH@0x00="k32w061dk6_ssbl.bin"
C:\NXP\DK6Programmer\DK6Programmer.exe -V2 -s %COMPORT% -P 1000000 -Y -p FLASH@0x00="ssbl_ext_flash_pdm_support.bin"

C:\NXP\DK6Programmer\DK6Programmer.exe -V5 -s %COMPORT% -P 1000000 -w image_dir_0=0000000010000000

C:\NXP\DK6Programmer\DK6Programmer.exe -V5 -s %COMPORT% -P 1000000 -w image_dir_1=00400000C9040101

C:\NXP\DK6Programmer\DK6Programmer.exe -V2 -s %COMPORT% -P 1000000 -Y -p FLASH@0x4000="chip-k32w0x-light-example.bin"
```

However, K32W serial message is wrong.
I changed MAC address to **.
[k32w serial log.txt](https://github.com/project-chip/connectedhomeip/files/11291720/k32w.serial.log.txt)
Can I resolve it?

marius-alex-tache, 2023-04-21:
What you are seeing is the tokenized log. You have to detokenize it using a python script. Another alternative would be to disable this tokenization, but you could run out of flash.

Please check the `Pigweed tokenizer` section from the README. You should find everything you need (including an example command) there. Let me know if it works.

mmt-10, 2023-04-21:
Thank you for your reply.

I forget `Pigweed tokenizer`.
I'm sorry.

I could get serial log.
Thank you very much for your support!

closed 2023-04-21 by mmt-10

marius-alex-tache, 2023-04-21:
You're welcome. Good luck!