project-chip/connectedhomeip #26045: can not pass the [TC-DD-2.1]

lisaaaas, 2023-04-11:
### Reproduction steps

step 14b,c execution failed and I'm not sure if I need to install some tools

### Bug prevalence

many times

### GitHub hash of the SDK that was being used

esp-idf

### Platform

esp32

### Platform Version(s)

1.0.2

### Type

Test Improvement

### Anything else?

[image]

lisaaaas, 2023-04-11:
@bzbarsky-apple

lisaaaas, 2023-04-11:
@vivien-apple

bzbarsky-apple, 2023-04-11:
dns-sd is only available on Darwin (MacOS/iOS/etc), and is always available there.

If you're on Ubuntu, you should be using avahi, and the verification steps should be listing the relevant avahi commands.

closed 2023-04-11 by bzbarsky-apple

lisaaaas, 2023-04-12:
@bzbarsky-apple @vivien-apple

bzbarsky-apple, 2023-04-12:
> but I also have a small problem with the instruction "'avahi-browse _matterc._udp -r" in Step 15b on ubuntu. I am not getting the expected results.

@lisaaaas What are you getting?  What do you expect to be getting?

lisaaaas, 2023-04-12:
@bzbarsky-apple

bzbarsky-apple, 2023-04-12:
@lisaaaas That does not answer my questions.  Again: What output are you getting (not the first line, the full output) and what output do you expect to be getting?

lisaaaas, 2023-04-13:
in Step 15b，My output are as follows：
[real log.txt](https://github.com/project-chip/connectedhomeip/files/11217346/real.log.txt)
The expected output is:
[expected  log.txt](https://github.com/project-chip/connectedhomeip/files/11217351/expected.log.txt)
According to the description of this test case:
[image]

lisaaaas, 2023-04-13:
@bzbarsky-apple

bzbarsky-apple, 2023-04-13:
OK, so which part is unexpected?  The spreadsheet gives an example, so obviously your hostname and instnace name and whatnot will be different, and the spreadsheet is listing the old formatting, and chip-tool has changed how exactly it prints things, in various ways (timestamp format is different, formatting of the log message prefix is different, the logging is obviously different, because you are running it on Linux and the spreadsheet is showing it running on Mac, and they have different dns-sd implementations with different logs, etc.

But the test is not saying "you must have this exact output", right?  That would be completely unreasonable; for example different devices _have_ to have different values for the hostname, and instance names are randomly generated.  What is the actual test condition that is supposed to be passing and is not @lisaaaas ?