project-chip/connectedhomeip #26047: [BUG] Light Switch App Setup is taking longer than expected

mishadesh, 2023-04-11:
### Reproduction steps

1. Build the Light Switch App for Nordic or Telink platform.
2. Flash the board.
3. Try to add device in Google Home. (used latest version - 3.1.1.1-dogfood)
4. On the latest step after choosing the device name the Setup is taking longer than expected will occur. The device will be commissioned to Thread network, but will not be paired and will not be accessible until factory reset.
[image]

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

03cb5e3d388ec242f06760734ce021a0ab20b4b8

### Platform

nrf, other

skoegugelfuss, 2023-04-21:
Same problem here.

@bzbarsky-apple I'm not sure this is on Google side... this >could< be an issue with the matter core SDK. I've found a reddit post about this. Here they say Google Home is waiting for a response for a short amount of time. If the response came to late, the Google Home app will give you that "Something went wrong" error...
Maybe there something went wrong in the matter core while setting up the device? :-D

skoegugelfuss, 2023-04-25:
There is an issue open on Google side.

Follow up here: https://issuetracker.google.com/issues/262773689?pli=1

zod6, 2023-10-05:
My experiences with GH and matter commissioning.
Tried with idf 5.1.1 and example/lighting-app from project-chip/connectedhomeip and espressif/esp-matter master branches, with two different phones, different google accounts and in different wifi networks.

steps to reproduce on ESP32-DevKitC and android 13 (latest updates installed):
A:
1. flash "light" from [launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml)
2. create new project and add matter device with vid=0xfff1 and pid=0x8000 on [developer console](https://console.home.google.com/) (otherwise it will show error "Not a Matter-certified device")
3. in same wifi network and under same google account, set up Google Nest Mini (without it there will be one extra warning page during commissioning step)
4. add wifi network using chip-tool: `pairing code-wifi 0x5164 myssid mypasswd 34970112332; pairing unpair 0x5164` (or in esp32 console `matter esp wifi connect myssid mypasswd`)
5. open Google Home app add matter device with pairing code [QR](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/_images/matter_qrcode_20202021_3840.png)
6. _device gets commissioned but after entering device name HA shows error 'Something went wrong'_

B (wifi not configured):
_steps 1-3 are same._
4.   open Google Home app add matter device with pairing code [QR](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/_images/matter_qrcode_20202021_3840.png)
5.   _device doesn't get wifi credentials and GH app gets stuck for 5+ minutes. logs from esp32:_
 ```
	I (13042) chip[NP]: ESP NetworkCommissioningDelegate: SSID: <ssid> // unaltered copy-paste
	...
	I (78932) chip[FS]: Fail-safe timer expired
	...
	E (78952) chip[SVR]: Commissioning failed (attempt 1): 32
```

I can commission and control ESP32 matter device with [Home Assistant](https://www.home-assistant.io) and [google home sample app](https://github.com/google-home/sample-apps-for-matter-android/releases/download/v1.4.1/GHSAFM-1.4.1-default-debug.apk) when wifi is configured