project-chip/connectedhomeip #26052: [BUG] Failed to get configured network when updating network status: Error ESP32:0x0500300F

vinay-apple, 2023-04-11:
### Reproduction steps

SHA: 0f1fc1b230eca4b4ce2f45c4a8cf1ce238284e96

Steps:

1. Trying to pair M5 board in Apple Home and pairing is failing with below errors seen on M5 logs (also attached)

E (22752) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
W (22762) wifi:Haven't to connect to a suitable AP now!
I (22762) chip[DL]: Attempting to connect WiFi station interface
E (22772) wifi:sta is connecting, return error
E (22772) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
W (22782) wifi:Haven't to connect to a suitable AP now!
E (22782) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F

[M5-pairing-fail.txt](https://github.com/project-chip/connectedhomeip/files/11201840/M5-pairing-fail.txt)

### Bug prevalence

Most of the times I attempt pairing 

### GitHub hash of the SDK that was being used

0f1fc1b230eca4b4ce2f45c4a8cf1ce238284e96

### Platform

esp32

### Anything else?

The issue is seen when trying to pair M5 on a multiple home setup where 2 Apple Homes on 2 different networks are prevalent and we try to pair M5 board on one of the Apple Homes.

bzbarsky-apple, 2023-04-11:
> `Error ESP32:0x0500300F`

That would be `ESP_ERR_WIFI_NOT_CONNECT`, I think.

From a bit earlier in the log:

> `I (24243) chip[NP]: ESP NetworkCommissioningDelegate: SSID: iPhone-12-pro-max`

@vinay-apple what sort of network is the network with that SSID?  Most importantly: does it have a 2.4GHz band?

rfortunatov, 2023-06-09:
Faced the same issue on ESP32C6 with latest esp-matter branch, though my case is different. I can successfully commission device to HA and Apple Home, though after approximately half an hour Wifi connection drops and devices enters a loop retrying the connection, after power-cycle it successfully connects to Wifi and operates for another half an hour or so.

rfortunatov, 2023-06-09:
@shubhamdp  Can this be ESP related rather than CHIP related?

karlp, 2023-06-09:
I've seen this, and needed to make sure that SPIRAM_TRY_ALLOCATE_WIFI_LWIP was enabled.  It worked without that on the plain "example-lock" but not by the time I had any sort of application active

rfortunatov, 2023-06-10:
@karlp Unfortunately there's no such option for Esp32-C6 module. It's supported only for plain Esp32 or Esp32-S3 modules. Or I'm terribly wrong and missing something.

rfortunatov, 2023-06-11:
I guess I've figured it out. 
What was bugging me with this issue is that my other unit that was flashed long ago was working just fine and the one flashed with latest esp-matter version was constantly failing. So, after retracing the steps it lead me to updating the IDF version from 420ebd208ae9eb71cb71ebd22742d1175f11addc to ea5e0ff298e6257b31d8e0c81435e6d3937f04c7 as a part of esp-matter update. Rolling back IDF version to 420ebd208ae9eb71cb71ebd22742d1175f11addc fixed it for me for good.

farhan-syed, 2024-06-25:
@rfortunatov What version is 420ebd208ae9eb71cb71ebd22742d1175f11addc? Having some issues for the past week trying to run Apple's matter example. Apples docs say to install IDF version 5.2.1.

dhrishi, 2024-10-25:
@Vaishaliavhale Please give it a try

>The issue is seen when trying to pair M5 on a multiple home setup where 2 Apple Homes on 2 different networks are prevalent and we try to pair M5 board on one of the Apple Homes.

VaishaliAvhale, 2024-10-25:
Hi @vinay-apple,

I reviewed the logs and found that the device failed to connect to WiFi. Please ensure that your iPad or iPhone is connected to a 2.4GHz network.

I tried commissioning the M5 board device using Apple Home, and it’s working fine on my end.

dhrishi, 2024-10-31:
I am closing this as we have verified. Please re-open in case this issue is still seen