project-chip/connectedhomeip #26053: [BUG] ZAP global attributes default to OPTIONAL

andy31415, 2023-04-11:
### Reproduction steps

Global attributes when processed by zap have the `isOptional` flag set. Global attributes in matter should not be optional.

When fixing, some double-checking of eventList may be required, as eventList is not implemented in 1.0 and 1.1, however spec does mark it as required.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

00a2dcc65c2d4075655622eea65311232d538435

### Platform

core

bzbarsky-apple, 2023-04-11:
@brdandu this seems like a ZAP issue?

andy31415, 2023-10-16:
Tested with latest zap, I did not see global attributes optional anymore.

We still need to special-case them since it looks like global attribute enabling respects zap (even though they are mandatory everywhere)

closed 2023-10-16 by andy31415