project-chip/connectedhomeip #26061: [BUG] [ESP32C3] [lighting-app] [OTA] matter command shows errors, but it can do OTA

mmt-10, 2023-04-12:
### Reproduction steps

Open new terminal (terminal 1).
```bash
cd ${IDF_PATH}
source export.sh
cd ${MATTER_SDK}
source scripts/activate.sh
cd examples/lighting-app/esp32/
idf.py set-target esp32c3
idf.py menuconfig
```

`CONFIG_ENABLE_OTA_REQUESTOR` is enabled on menuconfig.

```bash
idf.py build erase-flash flash
```

Open new terminal (terminal 2).
```bash
cd ${MATTER_SDK}
src/app/ota_image_tool.py create -v 0xFFF1 -p 0x8000 -vn 2 -vs "2.0" -da sha256 examples/lighting-app/esp32/chip-lighting-app.bin examples/lighting-app/esp32/chip-lighting-app.ota
./out/debug/chip-ota-provider-app -f examples/lighting-app/esp32/chip-lighting-app.ota
```
Open new terminal (terminal 3).
```bash
./out/debug/chip-tool pairing onnetwork 1 20202021
./out/debug/chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": null}]' 1 0
./out/debug/chip-tool pairing ble-wifi 2 ${WIFI_SSID} ${WIFI_PASSWORD} 20202021 3840
```
back to terminal 1.
```
minicom -D /dev/ttyUSB0

> matter ota query 1 1 0                                                        
                                                                                
Error: 2f
Error: 47 
```
I don't know above error codes.

back to terminal 3
```bash
./out/debug/chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 0 2 0
```
esp32-c3 can update via OTA.

What are Error `2F` and `47`?
How do I resolve it?

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

commit 4088a77f557e8571a39338fad51a1d8eb0131d79 (HEAD, tag: v1.0.0.2)

### Platform

esp32

bzbarsky-apple, 2023-04-12:
> I don't know above error codes.

https://github.com/project-chip/connectedhomeip/blob/master/docs/ERROR_CODES.md

0x2f is CHIP_ERROR_INVALID_ARGUMENT.

0x47 is CHIP_ERROR_INVALID_DESTINATION_NODE_ID.

>  How do I resolve it?

Presumably pass to "ota query" whatever arguments it actually expects?  Looks like `1 1 0` is not it.

mmt-10, 2023-04-12:
Thank you for your reply.
I understand error codes.

I followed below steps.
[https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/docs/guides/esp32/ota.md#query-for-an-ota-image](https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/docs/guides/esp32/ota.md#query-for-an-ota-image)

```
>matter ota query 1 <PROVIDER NODE ID> 0
```

As my understanding, `<PROVIDER NODE ID>` is `1` because I set it by below command.
```bash
./out/debug/chip-tool pairing onnetwork 1 20202021
```

bzbarsky-apple, 2023-04-12:
@mmt-10 It's entirely possible those directions are out of date.

bzbarsky-apple, 2023-04-12:
@mmt-10 In particular, it looks like "ota query" used to take 3 arguments before https://github.com/project-chip/connectedhomeip/pull/15248 and now takes no arguments.  It will return `CHIP_ERROR_INVALID_ARGUMENT` if it's passed more than 0 arguments.

mmt-10, 2023-04-13:
@bzbarsky-apple 
Thank you for your reply.
I understand it.

I remove arguments `1 1 0` , error disappear.
```
> matter ota query
```

closed 2023-04-13 by mmt-10