project-chip/connectedhomeip #26064: [ Question ] DRBG requirements and Crypto_DRBG() implementation

caipiblack, 2023-04-12:
Hi,

We have a question about  `Chapter 3.1. Deterministic Random Bit Generator (DRBG)` in the Matter specifications v1.0.0.

This is an extract of the chapter:

```
This protocol relies on random numbers for many security purposes. For example, random num­
bers are used in generating secret keys, counters, cryptographic signature generation random
secrets, etc. Those random numbers SHALL be generated using the Crypto_DRBG() function.

...

Crypto_DRBG() SHALL be implemented with one of the following DRBG algorithms as defined in
NIST 800-90A (the choice of which one is left to the manufacturer because the choice has no
impact on the interoperability):
```

I don't find any usages and implementations of `Crypto_DRBG()` in the connectedhomeip repository, so do you know which function should be re-implemented ? Is it `DRBG_get_bytes()` ?

Also, is this requirement automatically validated if we build the matter stack v1.0.0.2 on Linux with openssl backend (`CHIPCryptoPALOpenSSL.cpp`) ?

We are asking this question for certifications stuffs.

What we see is that `DRBG_get_bytes()` is implemented using `RAND_priv_bytes()` from openssl.

bzbarsky-apple, 2023-04-13:
> Is it DRBG_get_bytes() ?

Yes.

I don't know the answer offhand regarding the OpenSSL case....

caipiblack, 2023-04-13:
The [OpenSSL doc](https://www.openssl.org/docs/man3.0/man3/RAND_priv_bytes.html) about `RAND_priv_bytes()` is not clear:

```
NOTES
...
OpenSSL is configured to automatically seed the CSPRNG on first use using the operating systems's random generator.
```

Then I take a look on OpenSSL code and I found that there are default rand methods (called `drbg_xxx`)

The default rand methods in `crypto/rand/drbg_lib.c`:

```
RAND_METHOD rand_meth = {
    drbg_seed,
    drbg_bytes,
    NULL,
    drbg_add,
    drbg_bytes,
    drbg_status
};
```

This is confirmed by [this doc](https://www.openssl.org/docs/man1.1.1/man7/RAND_DRBG.html):

```
DESCRIPTION
The default OpenSSL RAND method is based on the RAND_DRBG class, which implements a deterministic random bit generator (DRBG). A DRBG is a certain type of cryptographically-secure pseudo-random number generator (CSPRNG), which is described in [NIST SP 800-90A Rev. 1].

While the RAND API is the 'frontend' which is intended to be used by application developers for obtaining random bytes, the RAND_DRBG API serves as the 'backend', connecting the former with the operating systems's entropy sources and providing access to the DRBG's configuration parameters.

Disclaimer
Unless you have very specific requirements for your random generator, it is in general not necessary to utilize the RAND_DRBG API directly. The usual way to obtain random bytes is to use [RAND_bytes(3)](https://www.openssl.org/docs/man1.1.1/man3/RAND_bytes.html) or [RAND_priv_bytes(3)](https://www.openssl.org/docs/man1.1.1/man3/RAND_priv_bytes.html), see also [RAND(7)](https://www.openssl.org/docs/man1.1.1/man7/RAND.html).
```

But OpenSSL also provide a mechanism to register default random generators at runtime. So at this point we can considere that if there is no "unsafe" random generator registered at runtime, the function `RAND_priv_bytes()` provides random bytes using DRBG.

So we pass this point of the certifications.

closed 2023-04-13 by caipiblack