project-chip/connectedhomeip #26065: [BUG] TC-RC-1.1 failing at Step 11

yinyihu-silabs, 2023-04-12:
### Reproduction steps

1. On your host machine, either linux/Mac, pull the SDK docker image that TH is using i..e docker pull connectedhomeip/chip-cert-bins:5a21d17fd2bb0a48f4a356937f4741f6c9dc7975
2. Launch the docker container using docker run -it -v <YOUR PAA TRUST STORE PATH>:/paa_roots -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket --privileged --network host -it connectedhomeip/chip-cert-bins:5a21d17fd2bb0a48f4a356937f4741f6c9dc7975
3. Run the RR script using rm -f admin_storage.json && python3 python_testing/TC_RR_1_1.py --discriminator $DISCRIMINATOR --passcode $SETUP_PASSCODE --commissioning-method on-network --paa-trust-store-path /paa_roots --storage-path admin_storage.json

### Bug prevalence

new to sha 5a21d17fd2bb0a48f4a356937f4741f6c9dc7975

### GitHub hash of the SDK that was being used

SVE1.1

### Platform

efr32, raspi

### Anything else?

Test runs with all-cluster-app from THv2.7 using latest version of TC-RR-1.1 script.
[th_test_rc_1_1_allclusters.log](https://github.com/project-chip/connectedhomeip/files/11211796/th_test_rc_1_1_allclusters.log)
[tc_rr_all_cluster.log](https://github.com/project-chip/connectedhomeip/files/11211797/tc_rr_all_cluster.log)

doublemis1, 2023-04-12:
@tehampson I think, thats not the issue. Based on test plan the minimal value of MaxGroupsPerFabric is 4, however it is for per group cluster, so if the device supports multiple Group clusters for example on endpoint 0, 1 (which is this case), the device should supports at least 8 MaxGroupPerFabrics, 4 per each group cluster.
```
Verify the value of MaxGroupsPerFabric read from the Group Key Management Cluster is no less than the Group Limits requirements of 4 (groups) * counted_groups_clusters.
```
where counted_groups_clusters is a number of supported Group clusters (on endpoint 0 and endpoint 1)

cjandhyala, 2023-04-12:
@tehampson Per spec the The node SHALL support at least three group keys per fabric. and The node SHALL support at least four groups per fabric per endpoint having a Groups cluster instance. 

[image]

@bzbarsky-apple based on your example , the device should support , 5 (max fabrics)*4 (max grps per fabric)*2 (two end pts) max grps

doublemis1, 2023-04-13:
@cjandhyala you are right, in my opinion, thus for that specific attribute (MaxGroupsPerFabric) should be at least 4 * 2 (endpoints supported group clusters 0, 1)

tehampson, 2023-04-13:
With the assumption that you are not setting `CHIP_CONFIG_MAX_GROUPS_PER_FABRIC` yourself, and the config is set for you by the SDK (from here):
```
#ifndef CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
#define CHIP_CONFIG_MAX_GROUPS_PER_FABRIC (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
#endif
```

and if there are only 2 group clusters enpoints for your all clusters application (I just know there are 3 when I compile `examples/all-clusters-app` for linux) then `CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC` should be set to 2. 

That should fix things for you, that way you will get the correct attribute read value in step 11 of this test.

PR #26106 (2023-04-14): [Silabs] Remove groups cluster from endpoint 0 on our lighting-app zap configs
fixes #26065

TC_RR_1.1 was failing on the silabs lighting app. `CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC` is set to 1 by default and our lighting app had the groups cluster on endpoint 0 and endpoint 1. 
You can define `CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC` to what your app needs, but more importantly, groups shouldn't be on endpoint 0. I believe it was used for some tests at the time.

This PR removes the groups cluster from endpoint 0.
closed 2023-04-14 by yufengwangca