project-chip/connectedhomeip #26073: [BUG] Unable to provision wi-fi in Lighting App via Bluetooth

javierbarellano4v, 2023-04-12:
### Reproduction steps

1. Launch Smart Home app in Android
2. Launch lighting-app on a Raspberry PI 4 running Ubuntu (with wi-fi enabled but not connected)
3. Proceed to add a matter device in Smart Home app.
5. App finds lighting app
6. Enter pairing code manually
7. Attempting to pair.
8. Lighting app shows it being paired.
9. Smart Home app show attempting to connect paired device with its current Wi-Fi network
10. Lighting app shows (see below) that the correct SSID from Smart Home app shows up in debug log.
11. Chip error (see log)

### Bug prevalence

each time during run

### GitHub hash of the SDK that was being used

NA

### Platform

raspi

### Anything else?

Here is an excerpt of the debug messages.

The SSID below has been obfuscated but it's the right network name.

1681311074.155897][14477:14477] CHIP:DMG: AccessControl: checking f=1 a=p s=0xFFFFFFFB00000000 t= c=0x0000_0031 e=0 p=a
[1681311074.155927][14477:14477] CHIP:DMG: AccessControl: implicit admin (PASE)
[1681311074.155964][14477:14477] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0006
[1681311074.156015][14477:14477] CHIP:NP: LinuxWiFiDriver: SSID: *********
**[1681311074.156055][14477:14477] CHIP:NP: Failed to connect to WiFi network: src/platform/Linux/ConnectivityManagerImpl.cpp:970: CHIP Error 0x00000003: Incorrect state**
[1681311074.156104][14477:14477] CHIP:DMG: Command handler moving to [ Preparing]
[1681311074.156136][14477:14477] CHIP:DMG: Command handler moving to [AddingComm]
[1681311074.156165][14477:14477] CHIP:DMG: Command handler moving to [AddedComma]
[1681311074.156184][14477:14477] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 1
[1681311074.156226][14477:14477] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1681311074.156388][14477:14477] CHIP:EM: <<< [E:47690r S:39927 M:130616833] (S) Msg TX to 1:FFFFFFFB00000000 [D3B3] --- Type 0001:09 (IM:InvokeCommandResponse)
[1681311074.156459][14477:14477] CHIP:IN: (S) Sending msg 130616833 on secure session with LSID: 39927
[1681311074.156682][14477:14477] CHIP:DMG: Command handler moving to [CommandSen]
[1681311074.156714][14477:14477] CHIP:DMG: Command handler moving to [AwaitingDe]
[1681311074.234196][14477:14478] CHIP:BLE: Indication confirmation, 0xffffa8035870
[1681311076.713276][14477:14478] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1681311076.713894][14477:14478] CHIP:BLE: Write request received debug 0xffffa8035870
[1681311079.275610][14477:14478] CHIP:BLE: Indication confirmation, 0xffffa8035870
[1681311081.753126][14477:14478] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1681311081.753448][14477:14478] CHIP:BLE: Write request received debug 0xffffa8035870

bzbarsky-apple, 2023-04-13:
@javierbarellano4v Does the Raspberry PI have `wpa_supplicant` installed?

javierbarellano4v, 2023-04-14:
I had done the configuration changes for wpa_supplicant as described in instructions in link but I had missed one file change which I did make and restarted the pi:

https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

Unfortunately, still get the same error.

Did I need to install something related prior to the configuration changes?

Thanks.

javierbarellano4v, 2023-04-14:
I replaced original ExecStart line in /etc/systemd/system/dbus-fi.w1.wpa_supplicant1.service with:

ExecStart=/sbin/wpa_supplicant -u -s -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf

Next,  in /etc/wpa_supplicant/wpa_supplicant.conf, I replaced with:

ctrl_interface=DIR=/run/wpa_supplicant
update_config=1

javierbarellano4v, 2023-04-14:
with regards to /etc/wpa_supplicant/wpa_supplicant.conf file, that's all i have in the file, the two lines in previous post.

javierbarellano4v, 2023-04-18:
Does this bug relate to https://github.com/project-chip/connectedhomeip/issues/23401 ?