project-chip/connectedhomeip #26079: [Feature] [nRF] Add Bridge example app demo

lboue, 2023-04-13:
### Feature description

Hello,

We already have the following examples:
[CHIP ESP32 Bridge App Example](https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/esp32)
[CHIP Linux Bridge Example](https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/linux)
[Matter Telink Bridge Example Application](https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/telink)

Will you add an bridge example app demo for nRF Connect soon?

### Platform

nrf

adminy, 2023-04-19:
Actually does that make sense? Since all nordic products have bluetooth and thread or wifi support, there is really no point in it being a bridge device as You can directly build examples with an nrf device. Bridge is for devices that have proprietary or non-matter communication protocols no? So I don't think it makes sense to focus on this.

lboue, 2023-04-20:
I would like to build a Zigbee bridge to Matter. I will need this example to start development. 
I don't think I'm the only one in this case.

Damian-Nordic, 2024-01-02:
@lboue We're not planning to contribute the bridge example to this repository, but if you're still interested, Matter to BLE bridge application is now part of our SDK: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/protocols/matter/overview/bridge.html. That application employs abstractions that may help replacing BLE with other transport protocols.

Please use https://devzone.nordicsemi.com/ for other requests or questions related to the bridge application and close this issue if that's OK with you :)

@adminy I think the bridge is a valid use case for manufacturers that have already developed non-Matter products that users might be more attracted to if they could be used with Matter ecosystems.

closed 2024-01-02 by lboue