project-chip/connectedhomeip #26083: [Android] commissioning flow

armysman, 2023-04-13:
Reproduction steps
1:Use Google Matter Home-Mobile SDK commission Matter device, get ip,port,passcode from callback.
2:Custom Fabric use  awaitEstablishPaseConnection(nodeId,ip,port,passcode) and awaitCommissionDevice(nodeId,NetworkCredentials) commission device
There are some doubts？
1: What parameters are required when Custom fabric init（ex：ChipDeviceController）
2: How to use ChipDeviceController.NOCChainIssuer API? I don't know how to integrate this in commissioning flow in Android app.
Thanks

andy31415, 2023-05-25:
@armysman I am missing some context here, could you adjust the description or provide more details?

- what are you trying to do?
- what guide(s) and SDK(s) are you using? based on the description I wonder if this should be routed to google-specific bits (which is not this repository but rather somewhere at https://developers.home.google.com/matter (Communities maybe)
- Could you provide some more concrete steps, using markdown features and potentially code snippets (formatted) to explain the issue better? I cannot follow/understand the current description.

armysman, 2023-06-14:
this issue solved my problems.thanks [https://github.com/project-chip/connectedhomeip/issues/27027](url)

closed 2023-06-14 by armysman