project-chip/connectedhomeip #26096: [BUG] Android Build Error

jeonghwan7, 2023-04-14:
### Reproduction steps

Android Build Error 

I think Because of this commit, happen to Android Build error.
https://github.com/project-chip/connectedhomeip/commit/5927a9adf9022cdb2a281aebab97fb06e8c7e21d#diff-9a593ef581c824078d9d89649dc54139c0952b20dc250a315cffd812b54cc6b4

Could you check it?

```
2023-04-14 13:57:57 INFO    /home/addy/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android24-clang++ -MMD -MF obj/src/access/libaccess.AccessControl.cpp.o.d -Wconversion -march=armv8-a -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/mbedtls/repo/include -c ../../src/access/AccessControl.cpp -o obj/src/access/libaccess.AccessControl.cpp.o
2023-04-14 13:57:57 INFO    clang++: error: unknown argument: '-fmacro-prefix-map=../../='
2023-04-14 13:57:57 INFO    [76/1381] c++ obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o
2023-04-14 13:57:57 INFO    FAILED: obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o 
2023-04-14 13:57:57 INFO    /home/addy/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android24-clang++ -MMD -MF obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o.d -Wconversion -march=armv8-a -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/mbedtls/repo/include -c ../../src/access/examples/PermissiveAccessControlDelegate.cpp -o obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o
2023-04-14 13:57:57 INFO    clang++: error: unknown argument: '-fmacro-prefix-map=../../='
2023-04-14 13:57:57 INFO    [77/1381] c++ obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o
2023-04-14 13:57:57 INFO    FAILED: obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o 
2023-04-14 13:57:57 INFO    /home/addy/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android24-clang++ -MMD -MF obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o.d -Wconversion -march=armv8-a -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/mbedtls/repo/include -c ../../src/access/examples/ExampleAccessControlDelegate.cpp -o obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o
```

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

62974a091b0366a18a8e88222121bbeea936dc9c

### Platform

android

bzbarsky-apple, 2023-04-14:
Duplicate of #25986

bzbarsky-apple, 2023-04-14:
@jeonghwan7 you need to use a newer version of the Android NDK.  See also https://github.com/project-chip/connectedhomeip/issues/25987

closed 2023-04-14 by bzbarsky-apple