project-chip/connectedhomeip #26097: [BUG] Android Provisioning Error using BLE

jeonghwan7, 2023-04-14:
### Reproduction steps

1. Launch Android App
2. Launch Lighting App Linux (Enable BLE and wpa_supplicant)
3. Select "PROVISION CHIP DEVICE WIFI"
4. Scan QR
5. Input WiFi Credentials 
6. Connecting to Lighting App in Linux
7. Connection Failed because of BLE Connection Error 

BLE Error Log is below in Android
```
04-14 14:25:17.192 D 2057     2278     BtGatt.GattService: onConnected() - clientIf=5, connId=0, status=133, address=001A7D_5 
04-14 14:25:17.193 D 5353     5537     BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=001A7D_5 
04-14 14:25:17.200 I 5353     5537     chip.BluetoothManager: MATTER-3840.onConnectionStateChange status = 133, newState=0 

```

v1.0.2 is well BLE provisioned, but the recent master code is not working well due to the above error code.
In my experience, if you try 10 times, it's about 2-3 times. I'm creating a ticket because I think I need to check it.

Log is attatched
[LogFilter_20230414_141853-BLEProvisioning-Error.zip](https://github.com/project-chip/connectedhomeip/files/11229082/LogFilter_20230414_141853-BLEProvisioning-Error.zip)

### Bug prevalence

2~3 / 10

### GitHub hash of the SDK that was being used

62974a091b0366a18a8e88222121bbeea936dc9c

### Platform

android

nicelyjust, 2023-05-16:
I am facing the same Issue，Do you have a reliable solution？ @jeonghwan7

jeonghwan7, 2023-06-09:
@nicelyjust Hi, Sorry for late reply!.
In my experience status=133 error is unstable network connection.
It seems that the success rate of commissioning increases if the commissioner and the commissioner are placed very close together.

nicelyjust, 2023-06-13:
Thank for your suggestion！ @jeonghwan7

yunhanw-google, 2023-11-06:
it seems it has been rooted caused with close distance.

closed 2023-11-06 by yunhanw-google