project-chip/connectedhomeip #26101: Thread implementation of ChipDnssdResolve does not follow the DNS-SD spec, breaks with some DNS servers

bzbarsky-apple, 2023-04-14:
It looks like the implementation of `ChipDnssdResolve` in `src/platform/OpenThread/DnssdImpl.cpp` ends up calling `otDnsClientResolveService`, which does queries for TXT and SRV records, but not the AAAA record.

DNS servers are allowed to include the AAAA record when an SRV record is queried (see https://datatracker.ietf.org/doc/html/rfc6763#section-12.2) but are not required to, and some (e.g. the Apple one) do not.

Devices are required to not assume the AAAA record will be included, see last paragraph of https://datatracker.ietf.org/doc/html/rfc6763#section-12 which says:

> Clients MUST be capable of functioning correctly with DNS servers (and Multicast DNS Responders) that fail to generate these additional records automatically, by issuing subsequent queries for any further record(s) they require.

But that assumption is _exactly_ what our `ChipDnssdResolve` implementation is making.  And when dealing with a DNS server that does not add the additional records, operational discovery will always fail.

What this means in practice is that OTA for the relevant Thread devices is unreliable at best (for example, it will work if the OTA provider and the node that sends the AnnounceOTAProvider command are the same node, but not otherwise), the sleepy check-in messages people are working on will just flat out not work in networks with a DNS server that does not add the additional records, binding will not work correctly in networks with such a DNS server, etc.

We need to either do the additional AAAA query when `otDnsClientResolveService` does not return an address, or fix `otDnsClientResolveService` to do that itself in addition to the TXT/SRV queries.

bzbarsky-apple, 2023-04-14:
@jwhui Would it be possible to make that change to `otDnsClientResolveService` in the "before this next Matter SDK release" timeframe?

bzbarsky-apple, 2023-04-14:
@jwhui Yes, we could fix this on the Matter side.  The question is whether `otDnsClientResolveService` should (optionally?) handle this for you.  Its current, not at all documented, behavior of sometimes providing the IP address and sometimes not is a bit confusing.  Other APIs I'm aware of either always provide the address (avahi, unless you explicitly pass the "don't resolve addresses" flag), or never provide it (the DNSService APIs on darwin).

jwhui, 2023-04-14:
@bzbarsky-apple , our goal with the OpenThread API was to provide the Additional Data when it is received, but if additional DNS queries are necessary (which has associated costs on a resource-constrained network like Thread), then leave that up to the caller. There may be reasons why the caller does not want to issue additional queries (e.g. it already has the information cached). We could probably improve the documentation to make this behavior more clear.

Abhayakara, 2023-04-14:
@jwhui I think this isn't really the right answer. There's no reason why otDNSClientResolveService() can't notice when it gets a response that the response doesn't contain address records in the additional section, and then do an additional query. Expecting individual application implementations to do this correctly seems prone to failure.

If there needs to be an API that doesn't fetch the records (which I agree makes sense), this should be non-default behavior, not default behavior. Otherwise we're going to have to individually validate every single Thread device, which feels like a Sisyphean task. We already know that we don't have an easy way to do a conformance test for this, so leaving this as the default behavior feels like playing with matches in a gas station.

abtink, 2023-04-14:
> The question is whether otDnsClientResolveService should (optionally?) handle this for you. Its current, not at all documented, behavior of sometimes providing the IP address and sometimes not is a bit confusing. 

@bzbarsky-apple, the behavior is documented and described in detail in OpenThread. Please see:

https://github.com/openthread/openthread/blob/f9cd4b4182d2fe6491ce161dd3ae297dad9c5847/include/openthread/dns_client.h#L567-L591

`otDnsClientResolveService()` performs a "Service Instance Resolution" and queries for SRV and/or TXT records (this follows definition in DNS-SD [RFC6763-section 5 "Service Instance Resolution"](https://www.rfc-editor.org/rfc/rfc6763#section-5)).
- If the server does provide AAAA record(s) in additional records section, we provide this in callback API (we allow the caller to iterate and get the full list of AAAA records if theare are multiple of them).
- If server does not, we indicate this, it is then up to callers/users of API to decide whether they want to do a address resolution for the host name or not. We provide seperate API `otDnsClientResolveAddress()` for this.

This design is intentional and gives flexibility to caller:
- We do not want to assume that user/caller is always interested in AAAA and do additional queries when not needed
    - User may not want to connect to service and dont need to do host name address resolution
    - Or user may previously resolved the address for the host and know the address.
- Always performing AAAA queriers on service resolution can be waseful
   - The OT DNS client is expected to run on all types of devices, including sleepy and battery-powered resource-constrained devices.

bzbarsky-apple, 2023-04-14:
> @bzbarsky-apple, the behavior is documented and described in detail in OpenThread. Please see:

What you linked to does not document whether `otDnsClientResolveService` will populate the relevant fields in the `otDnsServiceResponse`, which is the issue here.

> `otDnsClientResolveService()` performs a "Service Instance Resolution" and queries for SRV and/or TXT records (this follows definition in DNS-SD [RFC6763-section 5 "Service Instance Resolution"](https://www.rfc-editor.org/rfc/rfc6763#section-5))

The documentation at https://openthread.io/reference/group/api-dns#group__api-dns_1ga37875e2aa21b199dc0206099d60540eb does not link to the RFC and does not call out the possible pitfall here.

Importantly, we have existence proof that people as a result get things wrong.

> This design is intentional and gives flexibility to caller:

I agree that flexibility is useful, and the Avahi approach of just having a flag that says "don't do separate AAAA/A queries" provides that flexibility while having _default_ behavior that follows the RFC instead of pushing the understanding of the RFC compliance and pitfalls therein onto the API consumer.

Abhayakara, 2023-04-14:
It's generally just not good to assume that people will understand the protocol specification. Particularly people who don't do protocol work on that specific protocol as part of their day job. If you want to make sure that the application vendor only fetches the AAAA record if they need it, the way to do this would be to have a second asynchronous call to get the address record. This would be a no-op if we already have the address record, but would go fetch it if not. Now the app vendor only calls the API when they need the address record, because it's inconvenient, and it works correctly both in the case where the address record is in the additional data, and in the case where it is not.

Abhayakara, 2023-04-14:
In order for this to prevent mistakes, you'd have to take out the existing API call, though, which might be inconvenient.

abtink, 2023-04-14:
The issue here is that matter code used the `otDnsClient` APIs incorrectly. 
- Either misinterpreting the API behavior and/or assuming/expecting servers to always provide AAAA record in additional section.
- In my view, this should be addressed and fixed in matter code.

----

Independent of this, we can discuss whether it makes sense to add new APIs and/or extended the current OT APIs (e.g. adding flag that says "do AAAA query" on "Service Resolution", etc).

I strongly prefer to keep fewer and simpler APIs (keep it as is):
- `Browse()` queries for PTR, `ResolveService()` queries for SRV/TXT, and `ResolveAddress()` for AAAA/A.
- Adding more APIs and/or options/flags:
   - Makes it more complex for folks to follow and understand the APIs and their behavior.
   - Makes incorrect use more likely.
   - Adds more cases to be validated and tested.
   - Increases code size (impacts resource-constrained platforms that may not want to use certain features)
- With simple APIs, we provide all functions and full flexibility to users to handle all cases.

Also we do not generally change existing OT API behaviors (for backward compatibility).

We can argue that OT API documentation can be improved to make it more clear what is meant by "Service Resolution" and that it queries for SRV/TXT records and not directly for AAAA. 
- To me this is clear: We mention that AAAA record may not be available in "ServiceResolve" response in the callback functions, and also mention the SRV/TXT records explicitly in recently added [`otServiceMode` definition](https://github.com/openthread/openthread/blob/e63c5d321ad138f40fbca52afeb5c8d875fb445c/include/openthread/dns_client.h#L83-L98).
- But the fact that this was misinterpreted by matter developers is a good point that it may be helpful to clarify things even further.

----

On a related point, on `otDnsClientBrowse()` API if server provides SRV/TXT/AAAA records in the response (in additional section), we provide this info to the user from the callback and from the related `otDnsBrowseResponseGetServiceInfo()` functions.
-  It does not mean that we do query for SRV/TXT/AAAA records on `Browse()` call and will always have valid info for all services and these records. 
- We have this mentioned [here](https://github.com/openthread/openthread/blob/f9cd4b4182d2fe6491ce161dd3ae297dad9c5847/include/openthread/dns_client.h#L437-L439) (in browse callback).
- It may be useful to check that matter code is not assuming this when using `otDnsClientBrowse()` API.

----

> If you want to make sure that the application vendor only fetches the AAAA record if they need it, the way to do this would be to have a second asynchronous call to get the address record. 

The users of API can already do this and call `otDnsClientResolveAddress()` from the  `otDnsServiceCallback` (if they see there is no address which can be determined from the callback).

Abhayakara, 2023-04-14:
I feel like we're talking about two different things. I think we all agree that Matter is using the existing API incorrectly. I think we all understand what the API currently does. So there's no point in reiterating that.

The problem is that at present, it's less work to use the API incorrectly than it is to use it correctly, and an accessory developer will likely not realize this before they ship their product.

The question is whether it's worth changing the API to make it harder to use incorrectly than it is to use it correctly.

abtink, 2023-04-14:
> The problem is that at present, it's less work to use the API incorrectly than it is to use it correctly, and an accessory developer will likely not realize this before they ship their product.

If this is fixed in matter code, then I expect it will be used by matter accessory developers as well?

> an accessory developer will likely not realize this before they ship their product

Maybe this should be covered by cert tests in matter? 
- It is always possible for developers to use any (new) API incorrectly.
- It is better to validate behavior with tests than to try to protect against incorrect use through API changes

> The question is whether it's worth changing the API to make it harder to use incorrectly than it is to use it correctly.

We don't want to change existing OT APIs. Even if we add this behavior, it will be a new API and/or new config which accessory developers (and/or matter code) need to understand and use correctly.

- We may decide to add/extend OT APIs in future, but I don't want this to considered as a fix/solution here. 
- To me this is an issue in matter code and needs to be fixed in matter (matter stack may be used with older OT stack implementations and should use its current APIs correctly).

Abhayakara, 2023-04-15:
I'm not really concerned about Matter. Matter will be fine whether OpenThread addresses this issue or not. At this point we are assuming OpenThread will not accept a pull request to fix this issue, so it'll have to be fixed in the Matter SDK.

The reason I'm still talking about this is that I'm concerned about Thread. Having accessories fail randomly in ways that are hard to reason about and impossible for a developer to know to anticipate is bad for Thread as a whole, even if it has no effect on Matter.

bzbarsky-apple, 2023-04-15:
> It may be useful to check that matter code is not assuming this when using otDnsClientBrowse() API.

@abtink For browse, the semantics of the internal Matter "browse" API are that the callee may or may not return an IP address (unlike the internal "resolve" API, which is expected to always return an IP address).  So the Matter <-> OpenThread glue just does `otDnsClientBrowse` and then hands out whatever information it got, and the caller of that code will check whether it got an IP address.  If not, it will go ahead and do a "resolve" (which will be buggy as described above).

That said, this code seems to assume that if we got an IP address (i.e. got some AAAA records directly from the browse) then we _also_ got TXT records.  Based on the documentation we linked to, this seems like a bad assumption?

abtink, 2023-04-15:
> That said, this code seems to assume that if we got an IP address (i.e. got some AAAA records) directly from the browse then we also got TXT records. Based on the documentation you linked to, this seems like a bad assumption?

@bzbarsky-apple , correct. We should check to see what server provided in additional section of a browse  query response. 

We can use `otDnsBrowseResponseGetServiceInfo()` (from a `otDnsBrowseCallback`). The checks are [described here](
https://github.com/openthread/openthread/blob/e63c5d321ad138f40fbca52afeb5c8d875fb445c/include/openthread/dns_client.h#L432-L462), in particular for TXT record:
```- If no matching TXT record is found in @p aResponse, `mTxtDataSize` in @p aServiceInfo is set to zero.```

I don't think we can assume if server included SRV record (and AAAA record for related host) for a service instance, that it will also always include TXT record for it (in browse response).

kkasperczyk-no, 2023-04-18:
> @Damian-Nordic , is that something you can help take on?
We had an internal talk on that. I think that Abtin's hints regarding OT API are good enough, so I can try to fix it.

closed 2023-04-26 by bzbarsky-apple