project-chip/connectedhomeip #26104: [1.0] SkipChannel count should be SIGNED

andy31415, 2023-04-14:
### Reproduction steps

SkipChannel in channel-cluster is currently INT16U, however it should be signed to allow channels to move backwards.

This may mean also fixing some TV implementations

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

fcb34b7d410268259bb29177eb1223b9c035989b

### Platform

core

### Type

Spec Compliance Issue

bzbarsky-apple, 2023-04-14:
Note that this would be a backwards-incompatible change with the 1.0 SDK, which would cause commands sent by clients that have made this change to not be decodable at all by servers that have not been updated to the change, and vice versa.

The backwards-compatible thing is probably to change the spec to keep a field at index 0 which is unsigned int which only allows going up, and add a second, optional, field which allows going both up or down.  Or something.

How much of a worry that should be in practice, I can't tell you.

bzbarsky-apple, 2023-06-01:
Fixed in https://github.com/project-chip/connectedhomeip/pull/26501

closed 2023-06-01 by bzbarsky-apple