project-chip/connectedhomeip #26110: [Feature] Set accessory name in examples

lboue, 2023-04-15:
### Feature description

Hello,

Could you add the possibility to give a name to the accessories in the examples. The goal is to display this name during pairing instead of displaying "Matter accessory".

Regards

### Platform

all

sam-gabbay, 2023-04-17:
I am trying to do the same thing. I reached out to Nordic but they advised that I open a ticket here, they also mentioned it might be an Apple issue.

I am setting the NodeLabel in the ZapTool as well, but it is not coming up for me, only "Matter Accessory".

I believe this to also prevent other users in the Home (even if they are admins) to add these types of accessories. You need to be the "Owner" of the Home.

absoluteabutaj, 2023-04-19:
I believe the display name not preventing the device provisioning for the non-owner members in the home. Hopefully, it might be fixed in iOS/tvOS 16.5

sam-gabbay, 2023-04-28:
@bzbarsky-apple would you happen to have any ideas? I am setting the NodeLabel Attribute in the Basic Cluster as per Nordics recommendations, and I can set and read the values successfully, but in the Home app I am just getting "Matter Accessory".

bzbarsky-apple, 2023-04-28:
@sam-gabbay I don't know where the labeling is coming from exactly, but there's a nontrivial chance it comes from the DCL, not from the device itself.

absoluteabutaj, 2023-04-29:
@bzbarsky-apple - I checked with it and it's pulling from the DCL.

lboue, 2023-04-29:
I tried filling in the "Basic" cluster fields but it didn't change anything after factory reset an then pairing
[image]

lboue, 2023-04-29:
> I believe the display name not preventing the device provisioning for the non-owner members in the home. Hopefully, it might be fixed in iOS/tvOS 16.5

I am the owner members in the home. 
It does work for Philips Hue Matter bridge implementation so I note sure it is an iOS issue.

sam-gabbay, 2023-04-30:
@bzbarsky-apple can confirm as well, it is working with the DCL information.

@lboue setting the name must be done once the product / vendor is in the DCL apparently. Setting the Product Name / Label / NodeLabel will not do much for the display name at this point.

lboue, 2023-04-30:
I have already managed to fill in the global information of the accessory but:
* this does not appear in the name of the accessory in iOS
* this only appears in the characteristics of the accessory

[image]

Could you show me the rendering with iOS with the DCL?

herculesp17, 2025-01-21:
Any updates on this topic? I'd like to set the label for every single endpoint.