project-chip/connectedhomeip #26111: [BUG] unable to commission lighting-app demo on esp32 with iOS -- mbedTLS error: ECP - The signature is not valid

albrownwood, 2023-04-15:
### Reproduction steps

reproduction steps:
-clone this repo
-install build environments
-build examples/esp32/lighting-app/esp32
-flash example on ESPWROOM32 dev board
-start commissioning from Home app on iOS 16.4.1
-get Home error message: Unable to add accessory, pairing with accessory failed

see errors in accessory log attached

[log.txt](https://github.com/project-chip/connectedhomeip/files/11240456/log.txt)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

2e4064e17b2bfa8394100483279a4bbaf9d40b42

### Platform

esp32

### Platform Version(s)

ESPWROOM32, iOS 16.4.1

bzbarsky-apple, 2023-04-16:
@albrownwood Is this reliably reproducible?  If so, could you please can you please follow instructions below to file a feedback assistant ticket which would include logs from both the esp32 device and the iOS device?

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

bzbarsky-apple, 2023-04-16:
@albrownwood And if you do file the feedback assistant ticket, please mention the ticket number here so I can make sure it gets looked at.

albrownwood, 2023-04-20:
closing as I am not able to reproduce that issue

closed 2023-04-20 by albrownwood