project-chip/connectedhomeip #26117: Building the app is failing after disabling User label through zap tool

chirag-silabs, 2023-04-17:
### Reproduction steps

When we disable user label from zap tool and build the application, it is failing.

Issue:
FAILED: chip-efr32-lock-example.out chip-efr32-lock-example.out.map 
arm-none-eabi-g++ T../../../../examples/lock-app/silabs/efr32/third_party/connectedhomeip/examples/platform/silabs/efr32/ldscripts/efr32mg24.ld -Wl,defsym -Wl,SILABS_WIFI=1 -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,fatal-warnings -fdiagnostics-color -Wl,-gc-sections -Wl,-Map,./chip-efr32-lock-example.out.map @./chip-efr32-lock-example.out.rsp -o ./chip-efr32-lock-example.out
/home/test/stash/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/src/app/util/lock-common.util.cpp.o: in function `emberAfInit()':
/home/test/stash/matter/out/rs9116/lock_sleep_1/BRD4186C/../../../../examples/lock-app/silabs/efr32/third_party/connectedhomeip/src/app/util/util.cpp:108: undefined reference to `MatterUserLabelPluginServerInitCallback()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Log: 
[zap.log](https://github.com/project-chip/connectedhomeip/files/11246296/zap.log)

### Bug prevalence

5/5

### GitHub hash of the SDK that was being used

RC_2.0.0-1.1 - d03a56b7eeb06b25eab61196933d74ae61b4a4ea

### Platform

efr32

bzbarsky-apple, 2023-04-17:
@chirag-silabs What exact commands did you run after modifying the ZAP config?  It sounds like you did not regenerate the .matter file for the application, but it's hard to tell because you did not include the actual commands you ran, just the output from one of them.

chirag-silabs, 2023-04-18:
@bzbarsky-apple 
1. Ran zap tool app(<repo>: ./scripts/tools/zap/run_zaptool.sh). 
2. From the zap tool application: Open lock-app.zap file located in "matter_repo\examples\lock-common"  
3. Disable user label for endpoint 0
4. generate zap files and paste it in
        matter_repo\zzz_generated\lock-app\nxp\zap-generated
5. Build the application: 
Command used: `./scripts/examples/gn_efr32_example.sh examples/lock-app/silabs/efr32 out/rs911x_lock BRD4186C disable_lcd=true use_external_flash=false chip_enable_ble_rs911x=true --wifi rs9116`

Build used: 
Branch: SMG v2.0.0-1.1 - Build 12 
Commit: d03a56b7eeb06b25eab61196933d74ae61b4a4ea

bzbarsky-apple, 2023-04-18:
> generate zap files 

How _exactly_ was this part done?  I asked above for the exact commands, @chirag-silabs .

chirag-silabs, 2023-04-19:
@bzbarsky-apple  Can you please direct me to the resource to generate new zap/matter file with the required changes?

bzbarsky-apple, 2023-04-19:
@chirag-silabs and if you were using some other steps, based on some other documentation, it would be good to know what that is so we can fix it.

chirag-silabs, 2023-04-20:
Thanks @bzbarsky-apple. It is working now.
Actually, after changing the .zap file, we are generating the files in ZZZ_generated folder. May be our understanding was wrong.
I followed the steps which you had mentioned above . It is working fine now. Thanks for sharing the document.

closed 2023-04-20 by chirag-silabs

bzbarsky-apple, 2023-04-20:
@chirag-silabs You _can_ generate things in the zzz_generated folder but then you would need to use the pregen arguments to the build to actually use those, right?