project-chip/connectedhomeip #26127: Need YAML support for automatic generation of payload and PAKE-Verifier

manjunath-grl, 2023-04-17:
### Feature description

Summary:
          As some of the Semi-Automated testcases which uses OpenCommissioningWindow [ ECM ] it requires` PAKE-Verifier` to open commissioning window. And for TH2 commissioning process it requires the generated `Payload` value from the `Discriminato`r and `SetupPincode`. These value we want to be generated automatically in YAML.

Diff link: 

Affected tests:
TC-ACL-2.7
TC-ACL-2.8
TC-ACL-2.10
TC-ACE-1.5
TC-CADMIN-1.3
TC-CADMIN-1.5
TC-CADMIN-1.9
TC-CADMIN-1.11
TC-CADMIN-1.13
TC-CADMIN-1.15
TC-CADMIN-1.22
TC-CADMIN-1.24

### Platform

vscode, other

wessica, 2023-04-20:
Does anyone know how to generate PAKE-Verifier value that I can use it in the TH configuration when testing these TCs?

bzbarsky-apple, 2023-09-21:
@manjunath-grl Why do you want to generate these?  Why are you not OK with just using pre-generated values for the fixed iterations/discriminator/passcode that was selected?