project-chip/connectedhomeip #26134: [Feature] I want a smaller libchipcontroller.so for android

luobosu, 2023-04-18:
### Feature description

...
I have remove the symbol tables of libchipcontroller.so , but it still too large , about 6MB ,   Anyone can help me to optimize it ?

For android jni libs.

### Platform

android

joonhaengHeo, 2023-04-27:
https://github.com/project-chip/connectedhomeip/issues/25556

yunhanw-google, 2023-07-27:
@luobosu could you retry with latest code, this issue should be resolved. Since we decouple generated ChipCluster, please use generic IM Java/JNI API listed in https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L887, you could refer latest android chip-tool for the usage where we have used the generic IM Java/JNI API across. thanks

closed 2023-07-27 by yunhanw-google