project-chip/connectedhomeip #26148: I want to run "./chip-tool onoff toggle i 1" with i=1...5 ( without only one CLI)

viethoang171, 2023-04-18:
### Reproduction steps

I loop the main.cpp 50 times in home/connectedhomeip/examples/chip-tool/main.cpp:
/*=====================================================================*/
                                             Main.cpp
/*=====================================================================*/
#include "commands/common/Commands.h"
#include "commands/example/ExampleCredentialIssuerCommands.h"

#include "commands/discover/Commands.h"
#include "commands/group/Commands.h"
#include "commands/interactive/Commands.h"
#include "commands/pairing/Commands.h"
#include "commands/payload/Commands.h"
#include "commands/storage/Commands.h"

#include <zap-generated/cluster/Commands.h>
#include <zap-generated/test/Commands.h>
// ================================================================================
// Main Code
// ================================================================================
int main(int argc, char * argv[])
{
    ExampleCredentialIssuerCommands credIssuerCommands;
    Commands commands;
    registerCommandsDiscover(commands, &credIssuerCommands);
    registerCommandsInteractive(commands, &credIssuerCommands);
    registerCommandsPayload(commands);
    registerCommandsPairing(commands, &credIssuerCommands);
    registerCommandsTests(commands, &credIssuerCommands);
    registerCommandsGroup(commands, &credIssuerCommands);
    registerClusters(commands, &credIssuerCommands);
    registerCommandsStorage(commands);
    commands.Run(argc, argv);
    mark++;
    if(mark<50)
    return main(argc,argv);
}

### Bug prevalence

First device completed toggle, but other device ( other nodeId) didn't toggle

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/v1.0-branch

### Platform

esp32

### Platform Version(s)

ESP32 WROOM KIT

### Type

Test Improvement

### Anything else?

This is log of chip-tool:

[1681839455.623143][114395:114395] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1681839455.626248][114395:114395] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1681839455.626620][114395:114395] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1681839455.626853][114395:114395] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1681839455.627413][114395:114395] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-gykxl7)
[1681839455.627623][114395:114395] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1681839455.627691][114395:114395] CHIP:DL: NVS set: chip-counters/reboot-count = 9 (0x9)
[... 192 lines cut ...]
[1681839460.064514][114395:114395] CHIP:CTL: System State Initialized...
[1681839460.064542][114395:114395] CHIP:IN: UDP::Init bind&listen port=5550
[1681839460.064575][114395:114395] CHIP:IN: UDP::Init bound to port=5550
[1681839460.064585][114395:114395] CHIP:IN: UDP::Init bind&listen port=5550
[1681839460.064617][114395:114395] CHIP:IN: UDP::Init bound to port=5550
[1681839460.064626][114395:114395] CHIP:IN: TransportMgr initialized
[1681839460.064808][114395:114403] CHIP:DL: CHIP task running
[1681839460.064930][114395:114403] CHIP:TOO: Sending command to node 0x1
[1681839460.065118][114395:114395] CHIP:CTL: Shutting down the commissioner
[1681839460.065127][114395:114395] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1681839460.065159][114395:114395] CHIP:CTL: Shutting down the controller
[1681839460.065168][114395:114395] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1681839460.065237][114395:114395] CHIP:DMG: IM WH moving to [Uninitialized]
[1681839460.065245][114395:114395] CHIP:DMG: IM WH moving to [Uninitialized]
[1681839460.065254][114395:114395] CHIP:DMG: IM WH moving to [Uninitialized]
[1681839460.065263][114395:114395] CHIP:DMG: IM WH moving to [Uninitialized]
[1681839460.065273][114395:114395] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1681839460.065302][114395:114395] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1681839460.065427][114395:114395] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-tiyzdX)
[1681839460.065579][114395:114395] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1681839460.065606][114395:114395] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1681839460.065616][114395:114395] CHIP:DL: Inet Layer shutdown
[1681839460.065626][114395:114395] CHIP:DL: BLE shutdown
[1681839460.065636][114395:114395] CHIP:DL: System Layer shutdown
[1681839460.065667][114395:114395] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.h:212: CHIP Error 0x00000003: Incorrect state

bzbarsky-apple, 2023-04-18:
chip-tool when running in non-interactive mode might be shutting things down and then not restarting them again if you just try to call main() again, since it does not expect you to do that.

In particular, you are failing on `GetConnectedDevice` on a `CHIPDeviceController` when its state is not `State::Initialized`.  

You'll need to do some tracking of what exactly happens with controller lifetimes in this very odd setup.

closed 2023-04-18 by bzbarsky-apple

viethoang171, 2023-04-19:
Thank you! So how can I controll onoff 5 device and caculate their delay time without call main() again? And can you explain the information of --trace_file for me?
./chip-tool onoff toggle 1 1 --trace_file onoff.tx
--> So what information onoff.txt does this file have?
Best regards!

bzbarsky-apple, 2023-04-19:
"trace_file" is where a log of the message traffic gets saved.

> So how can I controll onoff 5 device and caculate their delay time without call main() again? 

What are you actually trying to do?  Are they 5 different devices?  Do you want to calculate the "delay time" including session establishment to each device?  Or are you trying to talk to the same device 5 times?

viethoang171, 2023-04-19:
I want to establish a benchmark systerm:
I controll onoff them with only 1 CLI ( examples I create a script as "./chip-tool onoff toggle 1-50 1" mean toggle nodeID from 1 to 50). So I run demo 5 devices. And I want to caculate "time delay" including session establishment to each device! 
Thank you!

bzbarsky-apple, 2023-04-19:
Is there a reason the "only one CLI" can't be a script that then invokes chip-tool multiple times?

If you absolutely need the functionality in chip-tool itself you will need to do some digging through the code to see how to make it do things it's currently not designed to do.

viethoang171, 2023-04-19:
Yes, it's necessary for me with " only one CLI", because I have to test them with a long time and others function. I have read the flow of code ( command ./chip-tool onoff toggle 1 1). I have 1 idea, after chip-tool process argv[2] ( "toggle" ), I will check how chip-tool save NodeId, to transport message, and I execute "for nodeId from 1 to 50). May be it logical? 
Thanks!

bzbarsky-apple, 2023-04-19:
It might be logical, if you also make sure that all the setup/teardown happens properly for that use case.  Again, chip-tool may not be set up right now to repeatedly set up and tear down controllers.