project-chip/connectedhomeip #26151: Error to read a Matter accessory from Homekit app to sample iOS application

muralivm, 2023-04-18:
### Reproduction steps

Commission ERF32 911 board into Apple home kit
Read the accessory into the sample application via Apple Home APIs
To perform operations with the accessory i am trying to convert the MTRBaseDevice but I am getting an error 
Could you please help me?  

### Bug prevalence

1

### GitHub hash of the SDK that was being used

fc0e4a2c698b06a6467a4969981e3dbd87a82063

### Platform

darwin

### Anything else?

Error(../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/controller/CHIPDeviceController.cpp:574: CHIP Error 0x0000002F: Invalid argument): Failure while trying to stop the pairing process
(lldb)

bzbarsky-apple, 2023-04-19:
>  i am trying to convert the MTRBaseDevice

What does "convert the MTRBaseDevice" mean?

> Error(../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/controller/CHIPDeviceController.cpp:574: CHIP Error 0x0000002F: Invalid argument): Failure while trying to stop the pairing process

That does not correspond to the listed SHA above (as in, that line does not have that error return on it in https://github.com/project-chip/connectedhomeip/commit/fc0e4a2c698b06a6467a4969981e3dbd87a82063).

I am guessing you are calling `deviceBeingCommissionedWithNodeID` or `getDeviceBeingCommissioned` for a node id that is not being commissioned...

muralivm, 2023-04-19:
I have commissioned the Matter accessory in Homekit
Using Apple Home APIs retrieved the HMaccessory.
HMaccessory has matterNodeID, which I am passing into the deviceBeingCommissionedWithNodeID.

So can you help me in how can pass nodeID and point me documentation for sample code as well

bzbarsky-apple, 2023-04-20:
> I have commissioned the Matter accessory in Homekit

Ok, great.  So it's commissioned, and hence not being commissioned anymore.

>  HMaccessory has matterNodeID, which I am passing into the deviceBeingCommissionedWithNodeID.

That will fail, since the device is _not being commissioned_.

> So can you help me in how can pass nodeID 

How to pass nodeID to what?  The matterNodeID on the HMaccessory (which I am surprised is exposed) is in the Home fabric, right?  Are you trying to talk to the device in the Home fabric?  Commission it into your own fabric?  Something else?

muralivm, 2023-04-20:
**I am trying to do is** 
I am creating my own fabric and passing matterNodeID so that my HMaccessory can act as MatterAccessory, I can read the clusters in the accessory. 

Can I achieve this functionality or I am doing something wrong?

How i can access the home fabric to my sample application, is this possible because Homekit API is not exposed? can you help me on understanding more about home fabric.

bzbarsky-apple, 2023-04-20:
> I am creating my own fabric 

In that case you will have your own node ids you assign, right?  And the matteNodeID of the HMaccessory is not relevant to your fabric at all.

muralivm, 2023-04-21:
Could you please help me with this 
How i can access the home fabric to my sample application, is this possible because Homekit API is not exposed? can you help me on understanding more about home fabric?

bzbarsky-apple, 2023-04-26:
@muralivm Please file a feedback assistant ticket with that question.  See https://developer.apple.com/bug-reporting/#feedback-assistant and https://feedbackassistant.apple.com/new-form-response and

* Choose iOS/iPadOS to file an issue
* Choose the area as Home App and fill in the details of the type of issue and the description

closed 2023-06-24 by woody-apple