project-chip/connectedhomeip #26159: [Feature] support multiple network interfaces in one Matter device

chshu, 2023-04-19:
### Feature description

...
The feature aims to support the use case that a Matter device could run over multiple network interfaces, like on both Wi-Fi and Thread networks. Two typical scenarios:

- A Camera works on Wi-Fi for video streaming, and on Thread for low power sensor data during sleep mode;
- A bulb supports both Wi-Fi and Thread, the consumer could choose to commission the bulb to Wi-Fi or Thread network, depends on whether there is Thread BR their home or not.

As described in the Matter spec: [Network Commissioning Cluster](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/NetworkCommissioningCluster.adoc#network-commissioning-cluster), a Matter device may has one or more network interfaces, each interface should has an instance of the Network Commissioning Cluster.

### Platform

all

bzbarsky-apple, 2023-04-19:
@chshu I believe this is supported already at least on Linux for the all-clusters app.  What is the actual scope of the feature work being talked about here?  Supporting it on specific platforms (which ones)?  Something else?

chshu, 2023-04-20:
When we tried to enable the feature on ESP platform, some changes are required in the platform mDNS implementation to make the Wi-Fi and Thread coexist in single firmware, the PR will come out soon. So I filed this issue to record the SDK changes.

Meantime, we tested the commissioner from ecosystems (Google/Apple), and found the feature is not supported yet. The Apps commission the device to the network interface indicated by endpoint 0, but both abort at some following commissioning steps.

@bzbarsky-apple Do you have any info about the support plan by Apple ecosystem?

bzbarsky-apple, 2023-04-20:
@chshu I can't comment on future releases.