project-chip/connectedhomeip #26175: Building CHIPtool for use in embedded devices and as cloud backend for web commissioning

MartyMcFlyInTheSky, 2023-04-20:
Our company had the idea to provide automatic provisioning not via smartphone but via website and embedded.

Here's our scenario:
- A user should be able to log into our cloud and use the browser API to read in the QR-code. The browser should then use the new [Web Bluetooth API](https://developer.mozilla.org/en-US/docs/Web/API/Web_Bluetooth_API) to transfer matter related info to the device. Now of course we don't want to implement this ourselves and maybe expose the CHIP API to our server in some way and receive the bluetooth datastream in response which we can forward through said browser API. Question: Is this feasible? What is the current status regarding browser-based commissioning?

- We plan to manage a lot of products via matter where commissioning via phone might become impractical. We thought there might be a possibility in having a so called "leader" device in a group of devices which is enabled to commission other new devices into the network. This would require CHIPtool to run on our embedded device. As far as I can see the tech stack associated with chip-tool is huge. Is there a pruned version that fits on a microcontroller

All information regarding these two topics is highly welcome! Thanks in advance!

bzbarsky-apple, 2023-04-20:
chip-tool is designed as a test tool, not a production controller.  To support its testing function, it (1) includes a lot of code that would never be used in a production controller, (2) is designed for ease of testing, not codesize, (3) stores private keys in insecure ways (files on disk).

You really should not use chip-tool for any sort of production use.

bzbarsky-apple, 2023-04-20:
@MartyMcFlyInTheSky I'm not aware of any complete production-quality controller implementations that are open-source, so far.