project-chip/connectedhomeip #26192: [BUG] IlluminanceMeasurement return  CHIP Error 0x00000026: Wrong TLV type, but TemperatureMeasurement is not

nmtoan91, 2023-04-21:
### Reproduction steps

Hi, 
I'm developing a bridge app based on the Linux project. After defining my bridged device as IlluminanceSensor with IlluminanceMeasurement,  I get ` CHIP Error 0x00000026: Wrong TLV type` when trying to read the measured value from the chip-tool. 

However, there is no problem when I treat my device as TemperatureSensor with TemperatureMeasurement even though the logics are the same.

Can you help me to classify why TemperatureMeasurement  is fine but the IlluminanceMeasurement is not?

Here is the log from chip-tool:

```
[1682077742.669592][172281:172283] CHIP:DMG: ReportDataMessage =
[1682077742.669604][172281:172283] CHIP:DMG: {
[1682077742.669613][172281:172283] CHIP:DMG:    AttributeReportIBs =
[1682077742.669628][172281:172283] CHIP:DMG:    [
[1682077742.669646][172281:172283] CHIP:DMG:            AttributeReportIB =
[1682077742.669665][172281:172283] CHIP:DMG:            {
[1682077742.669675][172281:172283] CHIP:DMG:                    AttributeDataIB =
[1682077742.669687][172281:172283] CHIP:DMG:                    {
[1682077742.669707][172281:172283] CHIP:DMG:                            DataVersion = 0x13d5b323,
[1682077742.669718][172281:172283] CHIP:DMG:                            AttributePathIB =
[1682077742.669730][172281:172283] CHIP:DMG:                            {
[1682077742.669742][172281:172283] CHIP:DMG:                                    Endpoint = 0x3,
[1682077742.669760][172281:172283] CHIP:DMG:                                    Cluster = 0x400,
[1682077742.669775][172281:172283] CHIP:DMG:                                    Attribute = 0x0000_0000,
[1682077742.669786][172281:172283] CHIP:DMG:                            }
[1682077742.669799][172281:172283] CHIP:DMG:
[1682077742.669833][172281:172283] CHIP:DMG:                            Data = 1,
[1682077742.669844][172281:172283] CHIP:DMG:                    },
[1682077742.669858][172281:172283] CHIP:DMG:
[1682077742.669874][172281:172283] CHIP:DMG:            },
[1682077742.669897][172281:172283] CHIP:DMG:
[1682077742.669907][172281:172283] CHIP:DMG:    ],
[1682077742.669924][172281:172283] CHIP:DMG:
[1682077742.669942][172281:172283] CHIP:DMG:    SuppressResponse = true,
[1682077742.669955][172281:172283] CHIP:DMG:    InteractionModelRevision = 1
[1682077742.669963][172281:172283] CHIP:DMG: }
[1682077742.670088][172281:172283] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0400 Attribute 0x0000_0000 DataVersion: 332772131
[1682077742.670126][172281:172283] CHIP:TOO: Response Failure: Can not decode Data
[1682077742.670211][172281:172283] CHIP:EM: <<< [E:51462i S:45325 M:262603494 (Ack:223034964)] (S) Msg TX to 1:0000000000000457 [1122] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1682077742.670242][172281:172283] CHIP:IN: (S) Sending msg 262603494 on secure session with LSID: 45325
[1682077742.670294][172281:172283] CHIP:EM: Flushed pending ack for MessageCounter:223034964 on exchange 51462i
[1682077742.670403][172281:172281] CHIP:CTL: Shutting down the commissioner
[1682077742.670424][172281:172281] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682077742.670436][172281:172281] CHIP:CTL: Shutting down the controller
[1682077742.670447][172281:172281] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1682077742.670457][172281:172281] CHIP:IN: SecureSession[0x7f76b400b770]: MarkForEviction Type:2 LSID:45325
[1682077742.670468][172281:172281] CHIP:SC: SecureSession[0x7f76b400b770]: Moving from state 'kActive' --> 'kPendingEviction'
[1682077742.670477][172281:172281] CHIP:IN: SecureSession[0x7f76b400b770]: Released - Type:2 LSID:45325
[1682077742.670490][172281:172281] CHIP:FP: Forgetting fabric 0x1
[1682077742.670506][172281:172281] CHIP:TS: Pending Last Known Good Time: 2023-04-06T15:30:05
[1682077742.670577][172281:172281] CHIP:TS: Previous Last Known Good Time: 2023-04-06T15:30:05
[1682077742.670587][172281:172281] CHIP:TS: Reverted Last Known Good Time to previous value
[1682077742.670610][172281:172281] CHIP:CTL: Shutting down the commissioner
[1682077742.670619][172281:172281] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682077742.670628][172281:172281] CHIP:CTL: Shutting down the controller
[1682077742.670637][172281:172281] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1682077742.670922][172281:172281] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1682077742.670954][172281:172281] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1682077742.670974][172281:172281] CHIP:FP: Shutting down FabricTable
[1682077742.670986][172281:172281] CHIP:TS: Pending Last Known Good Time: 2023-04-06T15:30:05
[1682077742.671042][172281:172281] CHIP:TS: Previous Last Known Good Time: 2023-04-06T15:30:05
[1682077742.671051][172281:172281] CHIP:TS: Reverted Last Known Good Time to previous value
[1682077742.671162][172281:172281] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Ji09Fq)
[1682077742.671401][172281:172281] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1682077742.671428][172281:172281] CHIP:DL: NVS set: chip-counters/total-operational-hours = 50 (0x32)
[1682077742.671439][172281:172281] CHIP:DL: Inet Layer shutdown
[1682077742.671448][172281:172281] CHIP:DL: BLE shutdown
[1682077742.671458][172281:172281] CHIP:DL: System Layer shutdown
[1682077742.671614][172281:172281] CHIP:TOO: Run command failure: src/lib/core/TLVReader.cpp:240: CHIP Error 0x00000026: Wrong TLV type
```

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

bzbarsky-apple, 2023-04-21:
> even though the logics are the same.

The MeasuredValue attribute (attribute 0) of Temperature Measurement is of type `int16` in the spec.

The MeasuredValue attribute (attribute 0) of Illuminance Measurement is of type `uint16` in the spec.

If you look at `src/lib/core/TLVReader.cpp` line 240, which is where the error is coming from, that's someone trying to read an unsigned integer from the TLV, but the type in the TLV is not an unsigned integer type.

@nmtoan91 What does your logic for producing the value look like?  Are you doing your own TLV encoding (e.g. via AttributeAccessInterface), or are you just implementing an external attribute callback and letting the SDK do its own TLV encoding?

nmtoan91, 2023-04-22:
@bzbarsky-apple I got it now. I must define UNIT16U for illuminance's measured value. 
Thank you very much,

closed 2023-04-22 by nmtoan91