project-chip/connectedhomeip #26197: [BUG] Unknown cluster 0x0000_0050 when ./chip-tool modeselect change-to-mode

nmtoan91, 2023-04-22:
### Reproduction steps

Hi,
I'm writing a dynamic bride app for modeselect.
The read attributes work fine but when I try to change mode, I got the error Unknown cluster 0x0000_0050 
Could you help me to solve this please.

Here are the related logs:
```
[1682173625.887639][1926889:1926889] CHIP:EM: Handling via exchange: 32905r, Delegate: 0x5586197db428
[1682173625.887686][1926889:1926889] CHIP:DMG: InvokeRequestMessage =
[1682173625.887696][1926889:1926889] CHIP:DMG: {
[1682173625.887703][1926889:1926889] CHIP:DMG:  suppressResponse = false,
[1682173625.887715][1926889:1926889] CHIP:DMG:  timedRequest = false,
[1682173625.887732][1926889:1926889] CHIP:DMG:  InvokeRequests =
[1682173625.887743][1926889:1926889] CHIP:DMG:  [
[1682173625.887752][1926889:1926889] CHIP:DMG:          CommandDataIB =
[1682173625.887762][1926889:1926889] CHIP:DMG:          {
[1682173625.887772][1926889:1926889] CHIP:DMG:                  CommandPathIB =
[1682173625.887783][1926889:1926889] CHIP:DMG:                  {
[1682173625.887798][1926889:1926889] CHIP:DMG:                          EndpointId = 0x3,
[1682173625.887808][1926889:1926889] CHIP:DMG:                          ClusterId = 0x50,
[1682173625.887819][1926889:1926889] CHIP:DMG:                          CommandId = 0x0,
[1682173625.887828][1926889:1926889] CHIP:DMG:                  },
[1682173625.887840][1926889:1926889] CHIP:DMG:
[1682173625.887849][1926889:1926889] CHIP:DMG:                  CommandFields =
[1682173625.887860][1926889:1926889] CHIP:DMG:                  {
[1682173625.887869][1926889:1926889] CHIP:DMG:                          0x0 = 65,
[1682173625.887887][1926889:1926889] CHIP:DMG:                  },
[1682173625.887896][1926889:1926889] CHIP:DMG:          },
[1682173625.887906][1926889:1926889] CHIP:DMG:
[1682173625.887914][1926889:1926889] CHIP:DMG:  ],
[1682173625.887926][1926889:1926889] CHIP:DMG:
[1682173625.887935][1926889:1926889] CHIP:DMG:  InteractionModelRevision = 1
[1682173625.887941][1926889:1926889] CHIP:DMG: },
[1682173625.887976][1926889:1926889] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0050 e=3 p=o
[1682173625.887993][1926889:1926889] CHIP:DMG: AccessControl: allowed
[1682173625.888003][1926889:1926889] CHIP:DMG: Received command for Endpoint=3 Cluster=0x0000_0050 Command=0x0000_0000
[1682173625.888014][1926889:1926889] CHIP:ZCL: Unknown cluster 0x0000_0050
[1682173625.888033][1926889:1926889] CHIP:DMG: Command handler moving to [ Preparing]
[1682173625.888046][1926889:1926889] CHIP:DMG: Command handler moving to [AddingComm]
[1682173625.888058][1926889:1926889] CHIP:DMG: Command handler moving to [AddedComma]
[1682173625.888074][1926889:1926889] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1682173625.888109][1926889:1926889] CHIP:EM: <<< [E:32905r S:9723 M:147846981 (Ack:264746493)] (S) Msg TX to 1:000000000001B669 [1122] --- Type 0001:09 (IM:InvokeCommandResponse)
[1682173625.888124][1926889:1926889] CHIP:IN: (S) Sending msg 147846981 on secure session with LSID: 9723
[1682173625.888247][1926889:1926889] CHIP:DMG: Command handler moving to [CommandSen]
[1682173625.888261][1926889:1926889] CHIP:DMG: Command handler moving to [AwaitingDe]
[1682173625.888316][1926889:1926889] CHIP:EM: >>> [E:32904r S:0 M:54231155 (Ack:236639180)] (U) Msg RX from 0:9B3CCC59CC70E4EC [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1682173625.888327][1926889:1926889] CHIP:EM: Found matching exchange: 32904r, Delegate: (nil)
[1682173625.888343][1926889:1926889] CHIP:EM: Rxd Ack; Removing MessageCounter:236639180 from Retrans Table on exchange 32904r
[1682173625.889779][1926889:1926889] CHIP:EM: >>> [E:32905r S:9723 M:264746494 (Ack:147846981)] (S) Msg RX from 1:000000000001B669 [1122] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1682173625.889799][1926889:1926889] CHIP:EM: Found matching exchange: 32905r, Delegate: (nil)
[1682173625.889813][1926889:1926889] CHIP:EM: Rxd Ack; Removing MessageCounter:147846981 from Retrans Table on exchange 32905r
```

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

bzbarsky-apple, 2023-04-22:
Sounds like an issue in either your use of dynamic-bridge or in the dynamic-bridge setup.  Hard to say more without knowing more about what exactly you are doing...

nmtoan91, 2023-04-22:
@bzbarsky-apple could you help me to check it, here is my dynamic bridge setup:
I use the same code in the example to deploy the bridged device. 
chip-tool can trigger my emberAfExternalAttributeReadCallback() function when I use "chip-tool modeselect read ...". 
 

```const EmberAfDeviceType gBridgeGenericCurtainDeviceTypes_[] = { { DEVICE_TYPE_MODE_SELECT_DEVICE, DEVICE_VERSION_DEFAULT }, 
                                                            { DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT } };
constexpr CommandId GenericAttrsIncomingCommands_[] = {
    app::Clusters::ModeSelect::Commands::ChangeToMode::Id,
    kInvalidCommandId,
};

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN( GenericAttrs_)
    DECLARE_DYNAMIC_ATTRIBUTE(ModeSelect::Attributes::Description::Id, CHAR_STRING, kDescriptorAttributeArraySize_, 0),        /* Measured Value */
    DECLARE_DYNAMIC_ATTRIBUTE(ModeSelect::Attributes::StandardNamespace::Id,ENUM16,2,0 ),        /* Measured Value */
    DECLARE_DYNAMIC_ATTRIBUTE(ModeSelect::Attributes::SupportedModes::Id,ARRAY,kDescriptorAttributeArraySize_,0 ), 
    DECLARE_DYNAMIC_ATTRIBUTE(ModeSelect::Attributes::CurrentMode::Id,INT8U,1,0 ), 
    DECLARE_DYNAMIC_ATTRIBUTE(TemperatureMeasurement::Attributes::FeatureMap::Id, BITMAP32, 4, 0),     /* FeatureMap */
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedGenericClusters_)
    DECLARE_DYNAMIC_CLUSTER(ModeSelect::Id, GenericAttrs_, GenericAttrsIncomingCommands_, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs_, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasicInformation::Id, bridgedDeviceBasicAttrs_, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER_LIST_END;
DECLARE_DYNAMIC_ENDPOINT(bridgedGenericEndpoint_, bridgedGenericClusters_);
DataVersion gGenericDataVersions_[ArraySize(bridgedGenericClusters_)];
```

bzbarsky-apple, 2023-04-23:
OK.  But what does the ZAP configuration for your device look like?   What does your generated `IMClusterCommandHandler.cpp` end up looking like?

Chances are, you don't have the relevant cluster enabled anywhere in your ZAP config, so the generated dispatch code does not know about it.  And unlike attribute reads, the dispatch code can't just work by walking the data structures the `DECLARE_DYNAMIC_CLUSTER` provide.

So you have two options: Either you need to implement a `CommandHandlerInterface` that handles the commands for this cluster on this endpoint, or if you want to use the normal SDK implementation you should have some endpoint defined in your ZAP config that enables those commands for this cluster.  Then in your application startup you can disable that endpoint to avoid actually exposing it....

nmtoan91, 2023-04-23:
@bzbarsky-apple thank you.
But, It sounds too advanced for someone like me who is new to this. I guess these processes should be implemented in the SDK, right?
I didn't touch the zap file in examples/bridged-app/linux. So, I need to append coresponding cluster ID manually?

nmtoan91, 2023-04-23:
@bzbarsky-apple 
I copied ModeSelect part from all-clusters.zap to my project. But I got an error:
Could you help me to reproduce it, please? 

```
FAILED: chip-bridge-app chip-bridge-app.map
g++ -O0 -fPIC -Werror -Wl,--fatal-warnings -Wl,-z,defs -fdiagnostics-color -Wl,--gc-sections -pie liblogging-stdio.a -Wl,-Map,./chip-bridge-app.map @./chip-bridge-app.rsp -o ./chip-bridge-app
/usr/bin/ld: obj/third_party/connectedhomeip/src/app/clusters/mode-select-server/bridge-common.mode-select-server.cpp.o: in function `(anonymous namespace)::ModeSelectAttrAccess::Read(chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder&)':
/home/nuc7/Work/connectedhomeip/examples/bridge-app/echonetlinux/out/debug/../../third_party/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:67: undefined reference to `chip::app::Clusters::ModeSelect::getSupportedModesManager()'
/usr/bin/ld: obj/third_party/connectedhomeip/src/app/clusters/mode-select-server/bridge-common.mode-select-server.cpp.o: in function `emberAfModeSelectClusterChangeToModeCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::ModeSelect::Commands::ChangeToMode::DecodableType const&)':
/home/nuc7/Work/connectedhomeip/examples/bridge-app/echonetlinux/out/debug/../../third_party/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:104: undefined reference to `chip::app::Clusters::ModeSelect::getSupportedModesManager()'
/usr/bin/ld: obj/third_party/connectedhomeip/src/app/clusters/mode-select-server/bridge-common.mode-select-server.cpp.o: in function `verifyModeValue(unsigned short, unsigned char)':
/home/nuc7/Work/connectedhomeip/examples/bridge-app/echonetlinux/out/debug/../../third_party/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:266: undefined reference to `chip::app::Clusters::ModeSelect::getSupportedModesManager()'
```

closed 2023-04-24 by nmtoan91

bzbarsky-apple, 2023-04-24:
> I guess these processes should be implemented in the SDK, right?

No, they need to be done in the app as needed.