project-chip/connectedhomeip #26198: [Documentation] Inconsistency Write Interaction writeResponses

Apollon77, 2023-04-23:
### Documentation issues

Hello,

for the Write Interaction the specification say that for successfull writes **no** AttributeStatusIB should be generated in the response. Nothing in the specs say so.

But the Chip tool is doing this:

```
[1682207977172] [20032:6108995] [DMG] WriteResponseMessage =
[1682207977172] [20032:6108995] [DMG] {
[1682207977172] [20032:6108995] [DMG] 	AttributeStatusIBs =
[1682207977172] [20032:6108995] [DMG] 	[
[1682207977172] [20032:6108995] [DMG] 		AttributeStatusIB =
[1682207977172] [20032:6108995] [DMG] 		{
[1682207977172] [20032:6108995] [DMG] 			AttributePathIB =
[1682207977172] [20032:6108995] [DMG] 			{
[1682207977172] [20032:6108995] [DMG] 				Endpoint = 0x0,
[1682207977172] [20032:6108995] [DMG] 				Cluster = 0x1f,
[1682207977172] [20032:6108995] [DMG] 				Attribute = 0x0000_0000,
[1682207977172] [20032:6108995] [DMG] 			}
[1682207977172] [20032:6108995] [DMG] 				
[1682207977172] [20032:6108995] [DMG] 			StatusIB =
[1682207977172] [20032:6108995] [DMG] 			{
[1682207977172] [20032:6108995] [DMG] 				status = 0x00 (SUCCESS),
[1682207977172] [20032:6108995] [DMG] 			},
[1682207977172] [20032:6108995] [DMG] 			
[1682207977172] [20032:6108995] [DMG] 		},
[1682207977172] [20032:6108995] [DMG] 		
[1682207977172] [20032:6108995] [DMG] 		AttributeStatusIB =
[1682207977172] [20032:6108995] [DMG] 		{
[1682207977172] [20032:6108995] [DMG] 			AttributePathIB =
[1682207977172] [20032:6108995] [DMG] 			{
[1682207977172] [20032:6108995] [DMG] 				Endpoint = 0x0,
[1682207977172] [20032:6108995] [DMG] 				Cluster = 0x1f,
[1682207977172] [20032:6108995] [DMG] 				Attribute = 0x0000_0000,
[1682207977172] [20032:6108995] [DMG] 				ListIndex = Null,
[1682207977172] [20032:6108995] [DMG] 			}
[1682207977172] [20032:6108995] [DMG] 				
[1682207977172] [20032:6108995] [DMG] 			StatusIB =
[1682207977172] [20032:6108995] [DMG] 			{
[1682207977172] [20032:6108995] [DMG] 				status = 0x00 (SUCCESS),
[1682207977172] [20032:6108995] [DMG] 			},
[1682207977172] [20032:6108995] [DMG] 			
[1682207977172] [20032:6108995] [DMG] 		},
[1682207977172] [20032:6108995] [DMG] 		
[1682207977172] [20032:6108995] [DMG] 	],
[1682207977172] [20032:6108995] [DMG] 	
[1682207977172] [20032:6108995] [DMG] 	InteractionModelRevision = 1
[1682207977172] [20032:6108995] [DMG] }
[1682207977172] [20032:6108995] [DMG]  
```

Additional background: Amazon Alexa requires currently (since they added Write to AccessControlCLuster acl) that status SUCCESS is sent on write responses. 

Resolution options: 
A: If chip tool is source of truth then the specifications is wrong.
B: If specification is right then Chip-Tool and Alexa is wrong
C: if sending SUCCESS is optional then specification AND Alexa is wrong ;-)

bzbarsky-apple, 2023-04-24:
@Apollon77 That's what the 1.0 spec says, but as you noted that does not match the SDK and does not match the intent.

This was being tracked in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6313 and was fixed in https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6314 back in February.  What the spec says right now (currently being balloted for the Spring 2023 release) in "8.7.3.3. Write Path Data Process" is:

> * Otherwise perform the following:
>   * The data field SHALL be changed to the data indicated with the path.
>   * An [AttributeStatusIB](file:///Users/bzbarsky/connectedhomeip-spec/build/html/index.html#_attributestatusib) SHALL be generated with the SUCCESS Status Code, with the Path field duplicating the path.

bzbarsky-apple, 2023-04-24:
And in particular, what the spec said was not compatible with either the 1.0 SDK shipped clients or servers, as you discovered, which is why the spec was adjusted in this case.

closed 2023-04-24 by bzbarsky-apple

Apollon77, 2023-04-26:
Thank you for these information, maybe it could also be an idea to release "specs errata" between bigger official spec releases for such "highly incompatible/breaking/unexpected" things like this?

bzbarsky-apple, 2023-04-26:
@Apollon77 Those changes still have to go through the normal ballot+IPR process... which they have not completed yet.