project-chip/connectedhomeip #26201: How to view MATTER messages [1.0]

NNAA0917, 2023-04-24:
### Reproduction steps

I would like to check the contents of messages between nodes, but I cannot because they are encrypted.
Is it possible to check the contents of the messages using Wiresharkt or other means?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

58cd5fd

### Platform

darwin

### Type

Security Issue

bzbarsky-apple, 2023-04-24:
https://github.com/CHIP-Specifications/matter-dissector is a Wireshark dissector for Matter, but that repository is not public.  

https://gitlab.com/wireshark/wireshark/-/merge_requests/9865 looks like it got merged recently, but does not support the decryption bits yet.

closed 2023-04-24 by bzbarsky-apple

NNAA0917, 2023-04-24:
Thank you for confirming.
Is there any software other than Wireshark that can confirm the contents of the message?

bzbarsky-apple, 2023-04-24:
If you control the communication endpoints, you could run them in tracing mode so they log the payloads....

Otherwise, nothing I am aware of.