project-chip/connectedhomeip #26202: [BUG] [NXP] [K32W] [lighting-app] commissioning - chip-tool cannot find k32w

mmt-10, 2023-04-24:
### Reproduction steps

chip-tool cannot find k32w061.

1. I use chip-tool of v1.0.0.2 [https://github.com/project-chip/connectedhomeip/tree/v1.0.0.2/examples/chip-tool](https://github.com/project-chip/connectedhomeip/tree/v1.0.0.2/examples/chip-tool).

    It is not [https://github.com/NXP/matter](https://github.com/NXP/matter)

2. Regarding [https://github.com/project-chip/connectedhomeip/issues/26043](https://github.com/project-chip/connectedhomeip/issues/26043), I could build k32w firmware and flash it to k32w061.

    k32w firmware was built from [https://github.com/NXP/matter/releases/tag/v1.0.0.2.1](https://github.com/NXP/matter/releases/tag/v1.0.0.2.1).

```bash
$ git log -1
commit 32ebde335bf15a4852f562f54c6703fafede3575 (HEAD, tag: v1.0.0.2.1, tag: K32W0_V1.0.0.2.1_Matter_based_v1.0.0.2_tag)
Author: Gabriel Couturier <gabriel.couturier@nxp.com>
Date:   Wed Feb 15 10:12:18 2023 +0100

    [common] Update ot-nxp and openthread submodule to GitHub/NXP
```

## Log
[k32w-log-20230424.txt](https://github.com/project-chip/connectedhomeip/files/11305927/k32w-log-20230424.txt)

[chip-tppl-20230424.txt](https://github.com/project-chip/connectedhomeip/files/11305929/chip-tppl-20230424.txt)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 4088a77f557e8571a39338fad51a1d8eb0131d79 (HEAD, tag: v1.0.0.2)

### Platform

k32w

### Platform Version(s)

commit 32ebde335bf15a4852f562f54c6703fafede3575 (HEAD, tag: v1.0.0.2.1, tag: K32W0_V1.0.0.2.1_Matter_based_v1.0.0.2_tag)

### Anything else?

* openthread border router : raspberry pi 4 + nordic nrf52840
  * This set is working with other Matter Thread Cluster

bzbarsky-apple, 2023-04-24:
@mmt-10 The chip-tool log shows it trying BLE commissioning and not discovering any commissionable Matter devices.

The k32w log shows a device coming up that never advertises itself as commissionable.  Note these lines in the k32w log:

```
[826[P][DIS]Updating services using commissioning mode 0
...
[834[P][DIS]Advertise commission parameter vendorID=65521 productID=32773 discriminator=3840/15 cm=0
```
(note the cm=0 bit there, which means "extended discovery enabled, but not currently commissionable").

> k32w firmware was built from https://github.com/NXP/matter/releases/tag/v1.0.0.2.1.

I suggest reporting the issue in that repository, since it seems like that firmware is not actually advertising itself as commissionable in your case.

mmt-10, 2023-04-24:
@bzbarsky-apple 
Thank you for your reply.

@marius-alex-tache suggested it.
[https://github.com/project-chip/connectedhomeip/issues/26043#issuecomment-1506632383](https://github.com/project-chip/connectedhomeip/issues/26043#issuecomment-1506632383)

NXP repository does not accept the issue.
[image]

marius-alex-tache, 2023-04-24:
Hi, the K32W0 lighting app example uses `User-intent commissioning flow`, which means you will have to press the `USER INTERFACE` button (I assume you are using a DK6) to start BLE advertising.

This information can be found in the corresponding [README](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0#bluetooth-le-advertising). Please go through it, as it provides all the information needed to set up an app successfully. Most likely the answers to all these questions can be found there. Please reach out if you have additional questions or if you find unclear information.

mmt-10, 2023-04-24:
Hi,
Thank you for your reply.
I misunderstood `USERINTERFACE`.
I pressed `SW2`.

closed 2023-04-24 by mmt-10