project-chip/connectedhomeip #26206: [Platform]

joonhaengHeo, 2023-04-24:
### Reproduction steps

I have a question regarding Darwin builds.

1. When building Darwin App (src/darwin), is there a Build Option used for libCHIP.a?
2. Is there anything I can do to build the library with minimal library size?

### Platform

darwin

### Type

Manually tested with SDK

bzbarsky-apple, 2023-04-24:
> When building Darwin App (src/darwin), is there a Build Option used for libCHIP.a?

It depends on exactly how you are building it, and exactly what you mean by "Build Option".

> Is there anything I can do to build the library with minimal library size?

The thing that would probably help the most here is stripping debug symbols from the built library.

All of this only matters if you're not using the system Matter.framework, of course.

joonhaengHeo, 2023-04-25:
@bzbarsky-apple 
Thank you for your response.

I'm looking for a way to reduce the size of the libCHIP.a image when building the Matter.Framework. Using this, we are trying to implement a Matter-supported iOS App.

Please ask if there are any additional places to check.

joonhaengHeo, 2023-04-25:
Additionally, can you tell me how to clear the debug symbol you mentioned?

bzbarsky-apple, 2023-04-25:
@joonhaengHeo If you are building your own Matter.framework, presumably see `src/darwin/Framework/chip_xcode_build_connector.sh` for the builds that build libCHIP.  Make sure you are not building a Debug configuration.  Past that, you could try setting `optimize_debug_level="s"` in the gn args in that script.

And for removing the debug symbols, `strip`, but which exact arguments depends on exactly what you are trying to do.

woody-apple, 2023-06-23:
Please re-open if you still have questions here.

closed 2023-06-23 by woody-apple