project-chip/connectedhomeip #26208: Dynamic Bridge Build Fail [BUG]

jinbiaochenstayhungry, 2023-04-24:
### Reproduction steps

the first time i issued (https://github.com/project-chip/connectedhomeip/issues/26203) .

and quickly i fix my problems as suggested to update checkout,that is very wonderful job of these people.

but i again encounter some  similar problems, i think developers may miss some small fix.

below is the steps:

1、update  chip project to latest commit de0dfcfbc42fe574774dc440a0e89a8884b28573
2、edit dynamic bridge example zap file,enable scenes and groups cluster server from zap gui.
3、 regenerate .matter file 
4、build dynamic bridge example. 
5、 gn gen out/mac 
6、ninja -C out/mac

### Bug prevalence

always

### GitHub hash of the SDK that was being used

de0dfcfbc42fe574774dc440a0e89a8884b28573

### Platform

darwin

### Anything else?

when ninja is building code generation, something wrong come out, seems there are two problems:
first is :
    some micro was not defined,for example:
 ZCL_GROUPS_CLUSTER_REVISION  and ZCL_SCENES_CLUSTER_REVISION.

second is:
   some type assign problem int other fields:
gen/bridge/Scenes.h:16:25: error: assigning to 'uint32_t' (aka 'unsigned int') from incompatible type 'Optional<chip::AttributeId>' (aka 'Optional<unsigned int>')
        attributeID = t.attributeID;
../../third_party/connectedhomeip/src/app/data-model/List.h:62:22: note: candidate template ignored: could not match 'const unsigned char[N]' against 'const std::vector<uint8_t>' (aka 'const vector<unsigned char>')

../../third_party/connectedhomeip/src/app/data-model/List.h:40:8: note: candidate function (the implicit copy assignment operator) not viable: no known conversion from 'const std::vector<AttributeValuePair>' to 'const chip::app::DataModel::List<const chip::app::Clusters::Scenes::Structs::AttributeValuePair::Type>' for 1st argument
struct List : public Span<T>
       ^
../../third_party/connectedhomeip/src/app/data-model/List.h:40:8: note: candidate function (the implicit move assignment operator) not viable: no known conversion from 'const std::vector<AttributeValuePair>' to 'chip::app::DataModel::List<const chip::app::Clusters::Scenes::Structs::AttributeValuePair::Type>' for 1st argument

detail log below:
In file included from gen/bridge/BridgeClustersImpl.h:4:
gen/bridge/Groups.h:17:138: error: use of undeclared identifier 'ZCL_GROUPS_CLUSTER_REVISION'
      mClusterRevision(chip::CharSpan("clusterRevision"), 65533, 0 | ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE), ZCL_INT16U_ATTRIBUTE_TYPE, 2, ZCL_GROUPS_CLUSTER_REVISION)
                                                                                                                                         ^
In file included from ../../main.cpp:50:
In file included from gen/bridge/BridgeClustersImpl.h:5:
gen/bridge/Scenes.h:16:25: error: assigning to 'uint32_t' (aka 'unsigned int') from incompatible type 'Optional<chip::AttributeId>' (aka 'Optional<unsigned int>')
        attributeID = t.attributeID;
                      ~~^~~~~~~~~~~
gen/bridge/Scenes.h:17:24: error: no viable overloaded '='
        attributeValue = t.attributeValue;
        ~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:584:13: note: candidate function not viable: no known conversion from 'DataModel::DecodableList<uint8_t>' (aka 'DecodableList<unsigned char>') to 'const std::vector<unsigned char>' for 1st argument
    vector& operator=(const vector& __x);
            ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:604:13: note: candidate function not viable: no known conversion from 'DataModel::DecodableList<uint8_t>' (aka 'DecodableList<unsigned char>') to 'std::vector<unsigned char>' for 1st argument
    vector& operator=(vector&& __x)
            ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:608:13: note: candidate function not viable: no known conversion from 'DataModel::DecodableList<uint8_t>' (aka 'DecodableList<unsigned char>') to 'initializer_list<std::vector<unsigned char>::value_type>' (aka 'initializer_list<unsigned char>') for 1st argument
    vector& operator=(initializer_list<value_type> __il)
            ^
In file included from ../../main.cpp:50:
[... 31 lines cut ...]
    vector& operator=(vector&& __x)
            ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:608:13: note: candidate function not viable: no known conversion from 'DataModel::DecodableList<Structs::AttributeValuePair::DecodableType>' to 'initializer_list<std::vector<clusters::ScenesCluster::AttributeValuePair>::value_type>' (aka 'initializer_list<clusters::ScenesCluster::AttributeValuePair>') for 1st argument
    vector& operator=(initializer_list<value_type> __il)
            ^
In file included from ../../main.cpp:50:
In file included from gen/bridge/BridgeClustersImpl.h:5:
gen/bridge/Scenes.h:51:28: error: no viable overloaded '='
      t.attributeValueList = attributeValueList;
      ~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~
../../third_party/connectedhomeip/src/app/data-model/List.h:62:22: note: candidate template ignored: could not match 'const chip::app::Clusters::Scenes::Structs::AttributeValuePair::Type[N]' against 'const std::vector<AttributeValuePair>'
    constexpr List & operator=(T (&databuf)[N])
                     ^
../../third_party/connectedhomeip/src/app/data-model/List.h:40:8: note: candidate function (the implicit copy assignment operator) not viable: no known conversion from 'const std::vector<AttributeValuePair>' to 'const chip::app::DataModel::List<const chip::app::Clusters::Scenes::Structs::AttributeValuePair::Type>' for 1st argument
struct List : public Span<T>
       ^
../../third_party/connectedhomeip/src/app/data-model/List.h:40:8: note: candidate function (the implicit move assignment operator) not viable: no known conversion from 'const std::vector<AttributeValuePair>' to 'chip::app::DataModel::List<const chip::app::Clusters::Scenes::Structs::AttributeValuePair::Type>' for 1st argument
In file included from ../../main.cpp:50:
In file included from gen/bridge/BridgeClustersImpl.h:5:
gen/bridge/Scenes.h:71:138: error: use of undeclared identifier 'ZCL_SCENES_CLUSTER_REVISION'; did you mean 'ZCL_SWITCH_CLUSTER_REVISION'?
      mClusterRevision(chip::CharSpan("clusterRevision"), 65533, 0 | ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE), ZCL_INT16U_ATTRIBUTE_TYPE, 2, ZCL_SCENES_CLUSTER_REVISION)
                                                                                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                                                                                         ZCL_SWITCH_CLUSTER_REVISION
../../include/GeneratedClusters.h:15:27: note: 'ZCL_SWITCH_CLUSTER_REVISION' declared here
static constexpr uint16_t ZCL_SWITCH_CLUSTER_REVISION                           = 1;

bzbarsky-apple, 2023-04-24:
@achaulk-goog

achaulk-goog, 2023-04-24:
Looks like optional<> was not properly handled in codegen, https://github.com/project-chip/connectedhomeip/pull/26229 should fix part of this

jinbiaochenstayhungry, 2023-04-25:
> > ZCL_GROUPS_CLUSTER_REVISION and ZCL_SCENES_CLUSTER_REVISION.
> 
> Sounds like those probably need to be added to `examples/dynamic-bridge-app/linux/include/GeneratedClusters.h`?

thanks for your hard work.
looks like need to add by me for every cluster revision in GeneratedClusters.h, by why  move this define to brige jinja  codegen ,so that no one have to do this manually definition.

bzbarsky-apple, 2023-04-25:
That's a question for @achaulk-goog

PR #27077 (2023-06-05): Delete dynamic bridge example
Dynamic bridge is not maintained/used and it seems to periodically create compile errors. Since it uses std a lot (std::vector, std::string), its interoperability with embedded-friendly constructs in `DataModel` is non-trivial to maintain.

Delete example without a maintainer.

Likely fixes #26208 by making it obsolete (no more dynamic bridge app)