project-chip/connectedhomeip #26214: Unable to commission with thread device using iOS matter framework[1.0]

NNAA0917, 2023-04-24:
### Reproduction steps

I want to control a thread-enabled matter device by iOS matter framework, but I don't know how to implement it. Does anyone know how to implement this?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

58cd5fb

### Platform

darwin

### Type

Test Improvement

bzbarsky-apple, 2023-04-24:
> I want to control a thread-enabled matter device by iOS matter framework, but I don't know how to implement it.

Assuming that you are trying to put the device in your own fabric:

1. You use the MatterSupport framework (see https://developer.apple.com/documentation/mattersupport) to request that the device be brought onto the IP network and get an onboarding payload for the device's newly-open commissioning window.
2. You use Matter.framework to commission the device into your fabric and then control it.  See the APIs on MTRDeviceController for creating commisioning sessions and commissioning devices.

What is the specific question here @NNAA0917 ?

NNAA0917, 2023-04-25:
Thank you for your response.
Sorry for the lack of clarity.
What I am trying to do is the following.

1. prepare thread border router (using raspberry pi)
2. prepare thread lightning device.
3. join the device to the thread

I don't know how to implement the part that corresponds to 3 in this.
(The iOS version of CHIPTool does not allow thread participation, so I am thinking of implementing this additionally.)

bzbarsky-apple, 2023-04-25:
@NNAA0917 Is this for an application that will go into the app store or for some local thing that is OK with installing developer profiles and whatnot?

bzbarsky-apple, 2023-04-26:
@NNAA0917 So for an actual application that goes into the app store, for step 3 you would be using MatterSupport.framework and implement https://developer.apple.com/documentation/mattersupport/matteradddeviceextensionrequesthandler/selectthreadnetwork(from:)

At that point you can return the extendedPANID of the network to join, or defaultSystemNetwork to have the network automatically selected.

For the "automatically selected" case, the thread BR setup will need to use THClient to call https://developer.apple.com/documentation/threadnetwork/thclient/3801788-storecredentialsforborderagent?language=objc so that the system knows about the thread network.

woody-apple, 2023-06-23:
Closing this, as this should be handled through feedback assistant, or as @bzbarsky-apple pointed out, please check the documentation above.

closed 2023-06-23 by woody-apple

qawitherev, 2023-08-21:
> Assuming that you are trying to put the device in your own fabric:
>
> 1. You use the MatterSupport framework (see https://developer.apple.com/documentation/mattersupport) to request that the device be [...]

@bzbarsky-apple before this i use the `MTRDeviceController.commissionNode(node, commissionParams)` to commission the device onto a fabric, but then i have to provide ssid and pwd inside the `commissionParams`. If I were to use MatterSupport to get my device into an IP network and Matter framework to commission it, do I use the same `MTRDeviceController.commissionNode(node, commissionParams)` inside the `commissionDevice()` request handler extension, but omit out the ssid and pwd inside the `commissionParams?`

bzbarsky-apple, 2023-08-28:
> do I use the same MTRDeviceController.commissionNode(node, commissionParams) inside the commissionDevice() request handler extension, but omit out the ssid and pwd inside the commissionParams?

Yes, exactly.  At that point you are doing on-network commissioning, so do not need to provide network credentials.

> however it seems that this function is not executed

Which function?  I would expect your `func commissionDevice(in: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws` to be called, and then in there you `setupCommissioningSession` using the passed-in `onboardingPayload` (after creating the controller if needed), etc.

qawitherev, 2023-08-29:
@bzbarsky-apple 
Turns out I have to use Matter Extension in order to do use MatterSupport. My bad man. Also, [this article](https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem) doesn't mention anything about the extension 👀.

bzbarsky-apple, 2023-08-29:
@qawitherev Does https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem#Set-the-principal-class not cover that?