project-chip/connectedhomeip #26217: [Android] write/invoke API is not working in ChipTool

jagadeesh-ayla, 2023-04-24:
### Reproduction steps

1. Commission ad device using Chiptool.
2. go to wild card
3. select write option
4. enter endpoint, cluster and attribute ids.
5. In the next screen enter value to write and click write.
Operation failing with 136 error 

### Platform

android

### Platform Version(s)

Latest

### Type

Manually tested with SDK

### Anything else?

I am trying to write on off control (Cluster id 6 with endpoint 1) tried to write onoff attribute with value 1/0 and also tried with true/false but no success.

Here is the logs.

FindOrEstablishSession: PeerId = [1:0000000000000008]
FindOrEstablishSession: No existing OperationalSessionSetup instance found
Found an existing secure session to [1:0000000000000008]!
OperationalSessionSetup[1:0000000000000008]: State change 1 --> 5
2023Got connected device pointer
IM write() called
WriteClient moving to [AddAttribu]
<<< [E:52443i S:47351 M:52662903] (S) Msg TX to 1:0000000000000008 [B4B4] --- Type 0001:06 (IM:WriteRequest)
(S) Sending msg 52662903 on secure session with LSID: 47351
WriteClient moving to [AwaitingRe]
2023-endAllActiveAnimators on 0xb40000760b7f0860 (RippleDrawable) with handle 0xb40000751b6e9da0
2023-Ime callback not found. Ignoring unregisterReceivedCallback. callbackId: 237035715
2023show(ime(), fromIme=true)
>>> [E:52443i S:47351 M:215402175 (Ack:52662903)] (S) Msg RX from 1:0000000000000008 [B4B4] --- Type 0001:07 (IM:WriteResponse)
Found matching exchange: 52443i, Delegate: 0xb4000075cb6e95a0
Rxd Ack; Removing MessageCounter:52662903 from Retrans Table on exchange 52443i
WriteClient moving to [ResponseRe]
WriteResponseMessage =
{
	AttributeStatusIBs =
	[
		AttributeStatusIB =
		{
			AttributePathIB =
			{
				Endpoint = 0x1,
				Cluster = 0x6,
				Attribute = 0x0000_0000,
			}
				
			StatusIB =
			{
				status = 0x88,
			},
			
		},
		
	],
	
	InteractionModelRevision = 1
}
WriteClient moving to [AwaitingDe]
WriteAttributesCallback ReportError is called
2023Report error for Endpoint 1, cluster 6, attribute 0: java.lang.IllegalStateException: IM Status: 136
2023write attribute Done
<<< [E:52443i S:47351 M:52662904 (Ack:215402175)] (S) Msg TX to 1:0000000000000008 [B4B4] --- Type 0000:10 (SecureChannel:StandaloneAck)
(S) Sending msg 52662904 on secure session with LSID: 47351
Flushed pending ack for MessageCounter:215402175 on exchange 52443i

jagadeesh-ayla, 2023-04-24:
Not sure if i am missing something here, Please suggest on the right way to do it.

bzbarsky-apple, 2023-04-24:
0x88 is UNSUPPORTED_WRITE.

That's because the OnOff attribute of the On/Off cluster is readonly; you can't write it.  Its value is changed via the various commands the cluster defines.

closed 2023-04-24 by bzbarsky-apple

jagadeesh-ayla, 2023-04-25:
The write operation in Android ChipTool has code below which requires endpoint, cluster id and attribute id, passing attributeid as 0 which is onoff. Is something else required to pass here, I am not finding the right way to do it and is not described anywhere.

val writeRequest = AttributeWriteRequest.newInstance(endpointId, clusterId, attributeId, tlvWriter.getEncoded(), version)
    deviceController.write(writeAttributeCallback,
                      ChipClient.getConnectedDevicePointer(requireContext(),
                      addressUpdateFragment.deviceId),
                      listOf(writeRequest),
                      timedRequestTimeoutMs,
                      imTimeoutMs)

bzbarsky-apple, 2023-04-25:
Again: the OnOff attribute _is not writable, per spec_.  You cannot change its value with an AttributeWriteRequest.