project-chip/connectedhomeip #26219: [Documentation] Contributing CLA form

tetedoie, 2023-04-24:
### Documentation issues

1. In the [contributing guidelines](https://github.com/project-chip/connectedhomeip/blob/master/CONTRIBUTING.md), the third bullet point in the [contributing as an open source contributor](https://github.com/project-chip/connectedhomeip/blob/master/CONTRIBUTING.md#contributing-as-an-open-source-contributor) says `Have signed the [Matter Working Group CLA]` but the link is pointing to the CLA as a document not as a form to sign (https://gist.github.com/clapre/65aa9fc63981da765039e0bb7e8701be). 

How do we actually sign the CLA as an open source contributor ? Is there is a CLA to sign for companies who are not CSA Matter working group members ?

bzbarsky-apple, 2023-04-24:
> How do we actually sign the CLA as an open source contributor ?

When you create PR, the automation creates a link to the signature form, I believe.

tetedoie, 2023-04-25:
Hi @bzbarsky-apple, thank you. I'll try and close the ticket if it works.

closed 2023-04-25 by tetedoie