project-chip/connectedhomeip #26236: Certifications from Chip-Cert tool[1.0]

SYESHASWINI, 2023-04-25:
### Reproduction steps

Hello, I have generated the PAA/PAI/DAC certificates following the steps from [here](https://github.com/CHIP-Specifications/connectedhomeip/tree/master/src/tools/chip-cert).

As I change the Validity of the certificates -valid-from "2020-10-15 14:23:43" the validation of certificates are getting failed. Can these certificates be used for any of the product irrespective of their specification, because while generating the certificates there is no information required related to vendor and product side.

### Bug prevalence

whenever I execute the command

### GitHub hash of the SDK that was being used

none

### Platform

esp32

### Type

Spec Compliance Issue

bzbarsky-apple, 2023-04-25:
@SYESHASWINI For a start, you should probably use https://github.com/project-chip/connectedhomeip/tree/master/src/tools/chip-cert, not the thing you linked to, which is literally years out of date.

If you still have issues when using the current chip-cert, please provide the exact steps that lead to the failure.

SYESHASWINI, 2023-04-26:
@bzbarsky-apple  I have generated PAA, PAI and DAC certificated which are validated without any error.

In the above DAC is only the product specific with PID. Is PAI Same for all the products for specific vendor, where PAA is same for every vendor.

While the generated the chip-root. chip-ICA by changing the validity the error appears as:
**Failed certificate chain validation: ../../src/credentials/CHIPCert.cpp:488: Error 0x0000004A.**

I just need to know whether the Generated PAA,PAI and DAC certificates are sufficient for the Matter certification test.

Thank You.

bzbarsky-apple, 2023-04-26:
> Is PAI Same for all the products for specific vendor, where PAA is same for every vendor.

@SYESHASWINI That's up to the vendor and PAA involved.

> I just need to know whether the Generated PAA,PAI and DAC certificates are sufficient for the Matter certification test.

I don't know; that's a question for the CSA certification people.

SYESHASWINI, 2023-04-27:
Okay @bzbarsky-apple, Thank You.

closed 2023-04-27 by SYESHASWINI