project-chip/connectedhomeip #26241: [Documentation]

liaonh, 2023-04-25:
### Documentation issues

how to use the following arguments?
--spake2p-verifier-base64 <PASE verifier as base64> 
--spake2p-salt-base64 <PASE salt as base64>  
--spake2p-iterations <PASE PBKDF iterations>

### Platform

other

closed 2023-04-25 by bzbarsky-apple

liaonh, 2023-04-26:
> > how to use the following arguments?
> 
> Those arguments affect the commissioning window that gets opened by default by some example apps. You can use them if you want to test a specific SPAKE2P prover + salt + iterations combination.
how to get spake2p_verifier and salt and Device Attestation Certificate, can you show me a example?

bzbarsky-apple, 2023-04-26:
> But how to generate SPAKE2P prover + salt + iterations ?

You pick a number for iterations; anything you want as long as it's at least 1000 and no more than 100,000, per spec.

You use anything you want for the salt; literally random bytes.  Again, per spec, at least 16 and at most 32 of them.

You use anything you want for the prover (this is the "passcode" in Matter setup payloads), as long as it's in the right rane and avoids the small set of illegal values.  See "5.1.1.6. Passcode" in the spec and "5.1.7.1. Invalid Passcodes".

Then you run all those bits through the algorithm in the spec to compute the verifier.  You can use the `spake2p` binary built from the bits in `src/tools/spake2p` to do that.  See `spake2p gen-verifier --help`.

>  Do I need a DAC (Device Attestation Certificate) to generate them?

No.

>  Can you show me a example( for example, for VendorID=0xfff1, ProductId=0x8001)?

None of those things above depend on the vendor id or the product id.

liaonh, 2023-04-26:
> You pick a number for iterations; anything you want as long as it's at least 1000 and no more than 100,000, per spec.
>
> You use anything you want for the salt; literally random bytes. Again, per [...]

Thank you very much. 
I  am working with  examples within the matter SDK(examples\bridge-app\linux and examples\lighting-app\linux ) .
How can I change the default PAI or DAC to my own?  
Where is the source code to change?

bzbarsky-apple, 2023-04-26:
>  How can I change the default PAI or DAC to my own?

`examples/platform/linux/AppMain.cpp` does:

```
SetDeviceAttestationCredentialsProvider(LinuxDeviceOptions::GetInstance().dacProvider);
```

so you need to either set your own dacProvider in the device options (see `examples/all-clusters-app/linux/main.cpp` for an example of that), or call `SetDeviceAttestationCredentialsProvider` with something you control after the `AppMain` bit has run.