project-chip/connectedhomeip #26243: [BUG]

palak-android, 2023-04-25:
### Reproduction steps

I have tried build APK from Android studio.

- Open app
- TAP on "Provision Chip Device with Wi-Fi"
- Scan QR code

Application get crashed.

**Crash Logs:**
```
2023-04-25 13:04:28.593 17011-17011 SetupPayloadParser      com.google.chip.chiptool             E  Cannot load library.
                                                                                                    java.lang.UnsatisfiedLinkError: dlopen failed: library "libSetupPayloadParser.so" not found
                                                                                                    	at java.lang.Runtime.loadLibrary0(Runtime.java:1077)
                                                                                                    	at java.lang.Runtime.loadLibrary0(Runtime.java:998)
                                                                                                    	at java.lang.System.loadLibrary(System.java:1661)
                                                                                                    	at chip.setuppayload.SetupPayloadParser.<clinit>(SetupPayloadParser.java:78)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleScannedQrCode$lambda$8(BarcodeFragment.kt:209)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$ZQcs77Yp5AxXQmBnFOw77e2FItM(Unknown Source:0)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$ZQcs77Yp5AxXQmBnFOw77e2FItM.run(Unknown Source:4)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:240)
                                                                                                    	at android.os.Looper.loop(Looper.java:351)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8380)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
2023-04-25 13:04:28.593 17011-17011 e.chip.chiptool         com.google.chip.chiptool             E  No implementation found for chip.setuppayload.SetupPayload chip.setuppayload.SetupPayloadParser.fetchPayloadFromQrCode(java.lang.String, boolean) (tried Java_chip_setuppayload_SetupPayloadParser_fetchPayloadFromQrCode and Java_chip_setuppayload_SetupPayloadParser_fetchPayloadFromQrCode__Ljava_lang_String_2Z)
2023-04-25 13:04:28.593 17011-17011 AndroidRuntime          com.google.chip.chiptool             D  Shutting down VM
2023-04-25 13:04:28.601 17011-17011 AndroidRuntime          com.google.chip.chiptool             E  FATAL EXCEPTION: main
                                                                                                    Process: com.google.chip.chiptool, PID: 17011
                                                                                                    java.lang.UnsatisfiedLinkError: No implementation found for chip.setuppayload.SetupPayload chip.setuppayload.SetupPayloadParser.fetchPayloadFromQrCode(java.lang.String, boolean) (tried Java_chip_setuppayload_SetupPayloadParser_fetchPayloadFromQrCode and Java_chip_setuppayload_SetupPayloadParser_fetchPayloadFromQrCode__Ljava_lang_String_2Z)
                                                                                                    	at chip.setuppayload.SetupPayloadParser.fetchPayloadFromQrCode(Native Method)
                                                                                                    	at chip.setuppayload.SetupPayloadParser.parseQrCode(SetupPayloadParser.java:18)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleScannedQrCode$lambda$8(BarcodeFragment.kt:209)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$ZQcs77Yp5AxXQmBnFOw77e2FItM(Unknown Source:0)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$ZQcs77Yp5AxXQmBnFOw77e2FItM.run(Unknown Source:4)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:240)
                                                                                                    	at android.os.Looper.loop(Looper.java:351)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8380)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)

```

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

de0dfcfbc42fe574774dc440a0e89a8884b28573

### Platform

android

yunhanw-google, 2023-07-27:
@palak-android I tried recently, and cannot see the crash, could you retry it using latest code? thanks

yunhanw-google, 2023-08-09:
@palak-android , please retry, thanks

yunhanw-google, 2024-01-14:
Close it now, Feel free to reopen it if this is still problematic, thanks

closed 2024-01-14 by yunhanw-google