project-chip/connectedhomeip #26250: [Documentation] transmit pin code over bluetooth

adminy, 2023-04-25:
### Documentation issues

This is more of a question if its possible at all.

Examples generate a pin code and a QR code, once you connect to the device via bluetooth, by pressing a physical button or any other trigger could also be taken as a form of authentication.

Is it possible to send the pin code over bluetooth right away when advertised?

### Platform

nrf

### Anything else?

At the moment in the home App, the minute you press the bluetooth device that's found, it instantly asks for code. Would this be a limitation on the apple side of things expecting a code or is that not going to be a problem as long as the credentials are provided over bluetooth?

bzbarsky-apple, 2023-04-25:
@adminy the way this works "in the wild is":

1. The device stores a "verifier" that can be used to verify a provided passcode, but does not store the passcode itself.
2. The passcode is present somewhere on the device packaging or physical device, in the form of a manual pairing code or QR code or both.
3. When you connect to the device via bluetooth, you need to present it with the passcode to prove to the device that you are the entity that has possession of the device.

In particular, the device cannot send "the pin code" over bluetooth to the commissioner, because the device does not know "the pin code".

bzbarsky-apple, 2023-04-25:
Oh, and the example apps do know the passcode, because they are not real devices with actual physical packaging.  But real devices are _forbidden_ to "know" the passcode, per spec.  They might have it stored on some separate hardware from the verifier so they can show a visible QR code on a screen, but exfiltrating the passcode from the device over the air is disallowed by the specification.

closed 2023-04-25 by bzbarsky-apple

adminy, 2023-04-25:
Its making the setup experience bad because you have to put in a different code and if you have to set up hundreds of devices, this is not fun. I understand that this is done for security reasons but in certain usecases its a hindrance, especially since bluetooth is only on for the duration of the setup and then its off.