project-chip/connectedhomeip #26252: [BUG] mode-select-server always includes on-off-server.h, even if DEPONOFF is '0'

eric-ocasio-nxp, 2023-04-25:
### Reproduction steps

Create an application with mode select server and no on-off cluster. Try to build and it will fail since EMBER_AF_ON_OFF_CLUSTER_SERVER_ENDPOINT_COUNT is not defined.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

Latest

### Platform

darwin

### Anything else?

I think the include of on-off-server.h should be gated, similar to what is done in emberAfModeSelectClusterServerInitCallback. I don't know if the same macro should be used, but the general idea is the same.

Dependency with On/Off is optional in the spec.

bzbarsky-apple, 2023-04-25:
Yes, this should probably be conditioned on `EMBER_AF_PLUGIN_ON_OFF`.

@eric-ocasio-nxp can you create a PR for that, since you have a way to test it?

eric-ocasio-nxp, 2023-04-26:
Done.

https://github.com/project-chip/connectedhomeip/pull/26263

PR #26263 (2023-04-27): ModeSelect: Gate inclusion of on-off-server.h based on whether an on-off cluster is present in the app
Gate inclusion of on-off-server.h based on whether there is an on-off cluster present in the application. This resolves #26252.
closed 2023-04-27 by andy31415