project-chip/connectedhomeip #26256: [bluez] Linux platform Google home does not connect to ble，BLE broadcast TLV content is out of sequence

xuhaoorvibo, 2023-04-26:
### Reproduction steps

1、linux platform
1.1、Compile the example/linux/lighting-app
1.2、Scan the QR code through the Google app after startup
1.3、Google home is not connected to BLE.
1.4、If modify the broadcasting order of ble through hcitool, place TLV 02 01 05 at the top, and the Google Home commission complete

2、Other platforms
2.1、Other platforms can complete the commission normally,but if the BLE TLV broadcast order of other platforms is modified and 02 01 05 is not placed in the first byte, the same phenomenon will occur as on the Linux platform

### Bug prevalence

It always appears

### GitHub hash of the SDK that was being used

4088a77f557e8571a39338fad51a1d8eb0131d79

### Platform

other

### Anything else?

[image]

tcarmelveilleux, 2023-04-27:
[image]

The spec implies that the order should be Flags AD first (which is the most common order), but there is no normative statements.

There are no certification cases making explicit changes of order:
- [TC-DD-2.2](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#322-tc-dd-22-discovery-by-commissioner-verification-dut---commissioner) doesn't mention it

Overall, this may be an interoperability issue, but there is there is an argument to be made for the understanding that the order is explicit in the spec, and that devices should not advertise a different order. It's possible this is impossible to achieve without changes for some devices, although unlikely.

smides-nest, 2023-04-30:
The intention was for the order in the spec to be explicit, hence the explicit indexing of the bytes in the Advertising PDU payload starting with 0.

leorozendaal, 2023-05-01:
> The intention was for the order in the spec to be explicit, hence the explicit indexing of the bytes in the Advertising PDU payload starting with 0.

please file a spec-issue to get such statement into the spec

smides-nest, 2023-05-01:
On closer read of the BT spec, it doesn't seem that there's a required order for the Advertising Data elements, so it could be an issue on our end. I'm asking some folks internally that are more familiar with BLE. Based on that outcome, we can update the spec text to state whether the exact ordering is necessary or not.

leorozendaal, 2023-05-02:
filed spec issue to track: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6566

smides-nest, 2023-07-04:
Since CHIP-Specifications/connectedhomeip-spec/issues/6566 has been addressed and closed, should this also be closed?

leorozendaal, 2023-07-04:
@smides-nest   I assume so - but leave it to you and colleagues to judge if there is anything else to address on this topic beyond the spec clarification in 6566