project-chip/connectedhomeip #26259: [BUG] Crash on QR code scanning

palak-android, 2023-04-26:
### Reproduction steps

- Open app
- TAP on "PROVISION CHIP DEVICE WITH THREAD"
- Scan QR code

Application get crashed every time,
**Logs:**
```
com.google.chip.chiptool             E  FATAL EXCEPTION: main
Process: com.google.chip.chiptool, PID: 8365
java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:581)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
Caused by: java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067) 
Caused by: chip.setuppayload.SetupPayloadParser$SetupPayloadException: src/setup_payload/java/SetupPayloadParser-JNI.cpp:57: Error 0x0000002F
	at chip.setuppayload.SetupPayloadParser.fetchPayloadFromQrCode(Native Method)
	at chip.setuppayload.SetupPayloadParser.parseQrCode(SetupPayloadParser.java:18)
	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleScannedQrCode$lambda$8(BarcodeFragment.kt:209)
	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$ZQcs77Yp5AxXQmBnFOw77e2FItM(Unknown Source:0)
	at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$ZQcs77Yp5AxXQmBnFOw77e2FItM.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8741)
	at java.lang.reflect.Method.invoke(Native Method) 
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571) 
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067) 

```

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

de0dfcfbc42fe574774dc440a0e89a8884b28573

### Platform

android

joonhaengHeo, 2023-04-27:
Please let us know which QRCode value you used to reproduce the issue accurately.

joonhaengHeo, 2023-04-27:
https://github.com/project-chip/connectedhomeip/pull/25958

joonhaengHeo, 2023-05-09:
The error seems to be caused by an invalid QRCode. By the way, it seems that the value you sent is not a QRCode, but a Manual Pairing Code. How did you test it?

yunhanw-google, 2023-07-27:
We have refactored out the SetupPayloadParser-JNI.cpp, and use pure kotlin version, is still the issue? @palak-android

yunhanw-google, 2024-01-18:
close it now, please try it using latest code, if the issue persist, feel freee to create a new issue. thanks

closed 2024-01-18 by yunhanw-google