project-chip/connectedhomeip #26271: [Optimization] Global lists are constructed an element at a time requiring multiple `AttributeReportIB`

tehampson, 2023-04-26:
Background: When performing a wildcard subscription to lighting example app we noticed that it can take a longer then expected.

Problem: There is a lot of extra time spent breaking up an sending multiple packets. What seemingly takes up a lot of unneeded space is there are multiple `AttributeReportIB` for each list such as `AcceptedCommandList`, `GeneratedCommandList`, and `AttributeList`. For n number of entries in the list there are n+1 associated AttributeReportIB included in the report

Some measurements:
* There are 27 report packets that need to be sent from Accessory, for a total of `26076 bytes`.
* It was found that all these `AttributeReportIB` for all the lists took a total of `16597 bytes`. This could be reduced to a theoretical `4574 bytes`
  * Assumptions for the theoretical value that was calculated:
    * Estimate of reduced is based on looking at the total size of `AttributeReportIB` when `REPLACE` comes in and adding 4 bytes for every entry after
    * Anything with `AttributeReportIB.AttributePathIB.ListIndex` populated or anything that had `TLV::kTLVType_Array` type for `AttributeReportIB.Data` was classified as a list.
    * Total initial sized of `AttributeReportIB` was was obtained by grabbing `reader.GetLengthRead()` before and after `reading AttributeReportIB` and calculating the delta.
    * All lists can be condensed into single `AttributeReportIB`.
* If this theoretical is anywhere close to correct we could go from 26076 bytes to 13537 bytes.

Solution: List that are under a certain length are build all together so that they can be sent in one `AttributeReportIB` instead of the multiple reports that adds overhead for each element in the array bloating up the report

bzbarsky-apple, 2023-04-27:
I've been thinking about this a bit for the last week or so.  Some thoughts:

* Generally speaking, we will reach the start of a new list at some point partway through a packet.  Currently we don't know how much space is left, but we could probably expose that.
* When we reach the start of a new list, we have several options for encoding it:
    1. What we do now.
    2. Encode a single REPLACE for as many entries as we can fit in what's left of this packet, then one ADD per entry after that in following packets until the list is done.
    3. Determine the list length and see whether it could fit entirely inside a packet.  If so, try to encode a single REPLACE for the whole list.  If that fails, finish up the current packet immediately, and encode the list as a single REPLACE at the start of the next packet.  For lists that might not fit inside a packet, we could still do option 2.
    
Option 3 is guaranteed to be fewer total bytes transferred than option 2, I think, but might involve more (smaller) packets depending on where the list boundaries fall compared to the packet boundaries.

I _really_ wish we could do APPEND with a list instead of ADD per entry; that would be the optimal encoding.  But that would involve a change in the encoding, bumping the interaction model revision, etc....

tcarmelveilleux, 2023-04-27:
If option 3 could somehow target the "small lists" that are *always* there (global attributes of all clusters), it would hit the majority of the egregious cases of today and cut down any subscription against a whole endpoint by a large factor

bzbarsky-apple, 2023-04-27:
https://github.com/project-chip/connectedhomeip/pull/26286 implements option 2.  We can also consider doing option 3 on top of that....

closed 2023-05-01 by bzbarsky-apple