project-chip/connectedhomeip #26274: [BUG]

uascent, 2023-04-27:
### Reproduction steps

I am using Matter Support Framework to commission my Device, but very frustrait, the comission always fail at the begin, the success rate is very low, I saw the IOS 16.5 Beta 3 Release note said already fixed some matter issues. but why the commission success rate is low.  any body know the reason?

the device log is here:
[device.log](https://github.com/project-chip/connectedhomeip/files/11339033/device.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

1.0.1

### Platform

darwin

### Platform Version(s)

16.5 Beta3

bzbarsky-apple, 2023-04-27:
@uascent Can you please follow instructions below to file a feedback assistant ticket? Please include the device log you gather along with the sysdiagnoses, so they can be correlated with each other, and put the resulting ticket id in this issue so I can follow up regarding what's going on.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

woody-apple, 2023-06-23:
Closing this, as this should be handled through feedback assistant.

closed 2023-06-23 by woody-apple

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/