project-chip/connectedhomeip #26281: [BUG] [codegen] Some appcluster fields are integer enum rather than referencing formal enum

turon, 2023-04-27:
### Reproduction steps

Example: Thermostat cluster

The following fields are defined as primitive (raw int) enums:

[systemMode](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L176) and [thermostatRunningMode](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L181)

Yet the enums are defined elsewhere in the file:

[ThermostatSystemMode](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L48) and [ThermostatRunningMode](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L61)

Ask: make enum fields reference the defining enum rather than using a primitive enum with no connecting reference.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5a68477

### Platform

core

turon, 2023-05-03:
> This is basically blocked on [CHIP-Specifications/connectedhomeip-spec#6249](https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6249) because we want our XML matching the spec, right?

Perhaps.  This issue is the general case beyond just FeatureMap.  Many appcluster enum fields exhibit as primitive types despite the enum itself being defined.

bzbarsky-apple, 2023-05-03:
Yes, and there is a set of spec issues tracking this.

Since things we add to the XML get shipped as API, and we are at a point where changing shipped API is hard to impossible, we should not be changing things in the XML unless we are 100% sure that will be the final state.  In cases where we are doing things not in the spec yet, we run the risk of the spec ending up with different naming, etc (and this has already happened; this is not a hypothetical).