project-chip/connectedhomeip #26288: [BUG] [Fan Control] Fan Mode can set 7

mmt-10, 2023-04-28:
### Reproduction steps

```bash
$ ./chip-tool fancontrol write fan-mode 10 ${NODE_ID} 1
$ ./chip-tool fancontrol read fan-mode ${NODE_ID} 1
```

### Log
> [1682647394.537201][89674:89676] CHIP:TOO:   fan mode: 7

fan-mode was set `7`.
As my understanding, FanMode should support range: 0 to 6 in Matter-1.0-Application-Cluster-Specification.
I suggest `FanMode > 6` should reject.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

bd17b9f52f41eeac5aaa74c1ecf3a76e06dd6958

### Platform

esp32

### Anything else?

I guess it does not depend on ESP32 platform.
it depends on Matter SDK.

bzbarsky-apple, 2023-04-28:
When I use https://github.com/project-chip/connectedhomeip/commit/bd17b9f52f41eeac5aaa74c1ecf3a76e06dd6958 and the all-clusters-app example as server, and I do `./chip-tool fancontrol write fan-mode 10 ${NODE_ID} 1` the response I get back is:
```
[1682655168492] [83064:939224] [DMG] 				status = 0x87 (CONSTRAINT_ERROR),
```
(as expected) and the value read returns does not change from whatever it was before I issued the write.

@mmt-10 which exact server application are you using?  Are you using a clean tree at SHA bd17b9f52f41eeac5aaa74c1ecf3a76e06dd6958 for the server?

mmt-10, 2023-04-28:
Thank you for your reply.

Please refer to below.
[https://github.com/espressif/esp-matter/issues/354#issuecomment-1527013340](https://github.com/espressif/esp-matter/issues/354#issuecomment-1527013340)

fan-mode : `7` is correct behavior?
I want to reject `out of range` request.

Regarding above result, it is different from your result.

jadhavrohit924, 2023-04-28:
@mmt-10 all-clusters-app on https://github.com/project-chip/connectedhomeip/commit/bd17b9f52f41eeac5aaa74c1ecf3a76e06dd6958 is working fine as mentioned by @bzbarsky-apple. We are not sure about the root cause yet. We will look into it.

closed 2023-04-28 by bzbarsky-apple