project-chip/connectedhomeip #26291: Interoperability Issues: How does the Thread SRP Client ensure that the expected AAAA Record is obtained

fuxiaoming-lumi, 2023-04-28:
### Reproduction steps

According to commit [fa3563a](https://github.com/project-chip/connectedhomeip/commit/fa3563a9ea15f9baa34b3cb4434e4131f5e33643) , I want to bring up an interoperability issue. How about Thread SRP server do include AAAA address in additional records but only link local address from query response. Whether this condition ( https://github.com/project-chip/connectedhomeip/blob/master/src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread.cpp#L2852 )is enough.
More Details: https://github.com/openthread/openthread/discussions/8967

This commit [fa3563a](https://github.com/project-chip/connectedhomeip/commit/fa3563a9ea15f9baa34b3cb4434e4131f5e33643) do solve my use case with apple ecosystem (for additional records does not carry AAAA address).

Update my test result with Nest Hub Gen2.
1.Thread Accessory(SRP Client) Query Request
```
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
```
2.Nest Hub(SRP Server)  Query Response with LLA
```
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Answers
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN, priority 0, weight 0, port 5540, target DCA632F3954A0000.default.service.arpa
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Additional records
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN, addr fe80::dea6:32ff:fef3:954a
        Name: DCA632F3954A0000.default.service.arpa
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
        Time to live: 10 (10 seconds)
        Data length: 16
        AAAA Address: fe80::dea6:32ff:fef3:954a
```
3.Let Thread Accessory Query Request for AAAA
```
Queries
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN
        Name: DCA632F3954A0000.default.service.arpa
        [Name Length: 37]
        [Label Count: 4]
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
```
4. Nest Hub(SRP Server)  Query Response with ULA
```
Queries
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN
Answers
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN, addr fddc:3740:f4b5:4ff4:dea6:32ff:fef3:954a
        Name: DCA632F3954A0000.default.service.arpa
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
        Time to live: 10 (10 seconds)
        Data length: 16
        AAAA Address: fddc:3740:f4b5:4ff4:dea6:32ff:fef3:954a
```
5.Network is Reachable now

So currently there are maybe some options:
1. Add some requirements and mention in Thread Spec for SRP Server, I believe this needs to be mandatory, otherwise there will be interoperability issues. At least for Matter over Thread devices using SRP to  register their services it should not be a case, because they're assigning only "external" IPv6 addresses to the SRP host: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread.cpp#L2447 External then propagates to the GUA addresses only: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread.cpp#L1006
 But even if for some reason the device would receive only link local IP address what should we do? Using this IP address directly will result in not being able to reach the destination and querying for AAAA record again may end with returning exactly the same address.
2. Query again? Or whether it should be informed to the application layer to allow the device maker to make a decision.

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

e62d3ddaeeaed47535060f9d542ef87c655fef97

### Platform

core

### Platform Version(s)

1.0

Abhayakara, 2023-04-28:
It's not clear what you're asking for here.

fuxiaoming-lumi, 2023-04-28:
@Abhayakara According the logic from commit [fa3563a](https://github.com/project-chip/connectedhomeip/commit/fa3563a9ea15f9baa34b3cb4434e4131f5e33643)，srp client will not further query about AAAA record AAAA if the previous query response contains AAAA record. 

So the question is about if the AAAA address obtained in the previous query response was only the link local address. But only get the link local address(outside of thread network) will cause the network to be unreachable, right?

Network Topology:
```
raspberry pi (11DA090384C9CFB4-0000000000000001._matter._tcp.local)<------> WiFi Router <------> nest hub gen 2 <------> matter over thread accessory
```

If we follow the logic of this commit [fa3563a](https://github.com/project-chip/connectedhomeip/commit/fa3563a9ea15f9baa34b3cb4434e4131f5e33643)，the result would be as following.

a.Thread Accessory(SRP Client) Query Request
```
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
```
b.Nest Hub(SRP Server) Query Response with LLA
```
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Answers
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN, priority 0, weight 0, port 5540, target DCA632F3954A0000.default.service.arpa
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Additional records
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN, addr fe80::dea6:32ff:fef3:954a
        Name: DCA632F3954A0000.default.service.arpa
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
        Time to live: 10 (10 seconds)
        Data length: 16
        AAAA Address: fe80::dea6:32ff:fef3:954a
```
c. Network is unreachable

So I added a logic that if the received address is a LLA, then send additional DNS query to obtain IPv6 address for ULA. Here is my test result.
a.Thread Accessory(SRP Client) Query Request
```
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
```
b.Nest Hub(SRP Server) Query Response with LLA
```
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Answers
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN, priority 0, weight 0, port 5540, target DCA632F3954A0000.default.service.arpa
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Additional records
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN, addr fe80::dea6:32ff:fef3:954a
        Name: DCA632F3954A0000.default.service.arpa
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
        Time to live: 10 (10 seconds)
        Data length: 16
        AAAA Address: fe80::dea6:32ff:fef3:954a
```
c.Let Thread Accessory Query Request for AAAA
```
Queries
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN
        Name: DCA632F3954A0000.default.service.arpa
        [Name Length: 37]
        [Label Count: 4]
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
```
d.Nest Hub(SRP Server) Query Response with ULA
```
Queries
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN
Answers
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN, addr fddc:3740:f4b5:4ff4:dea6:32ff:fef3:954a
        Name: DCA632F3954A0000.default.service.arpa
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
        Time to live: 10 (10 seconds)
        Data length: 16
        AAAA Address: fddc:3740:f4b5:4ff4:dea6:32ff:fef3:954a
```
e.Network is Reachable now

So there are some issues left over here:
1. If for some reason the device would receive only link local address what should we do? Query again？But maybe still return the same address only LLA.
2. What is the best practice? Anyway, we need to consider this situation and solve this issue. Otherwise, I believe this will affect the related function such as ICD, Binding, etc

Abhayakara, 2023-04-28:
It’s an error for you to get a link local address that doesn’t work. How did this happen?

fuxiaoming-lumi, 2023-04-28:
```
raspberry pi (11DA090384C9CFB4-0000000000000001._matter._tcp.local)<------> WiFi Router <------> nest hub gen 2 <------> matter over thread accessory  

raspberry pi info:
[11DA090384C9CFB4-0000000000000001._matter._tcp.local]
[fe80::dea6:32ff:fef3:954a]
[fddc:3740:f4b5:4ff4:dea6:32ff:fef3:954a]
```

As shown in the network topology diagram, A Matter Thread Device attempts to resolve `11DA090384C9CFB4-0000000000000001._matter._tcp.local` and send out dns query for SRV & TXT.
```
Matter Thread Device --------> Nest Hub Gen2 (Sniffer Log No.32117)
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
```

Nest hub as the SRP server to answer this query include the Additional records but only have LLA. __So that's why the device received the LLA.__

```
Matter Thread Device <-------- Nest Hub Gen2  (Sniffer Log No.32127)
Queries
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Answers
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN, priority 0, weight 0, port 5540, target DCA632F3954A0000.default.service.arpa
    11DA090384C9CFB4-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
Additional records
    DCA632F3954A0000.default.service.arpa: type AAAA, class IN, addr fe80::dea6:32ff:fef3:954a
        Name: DCA632F3954A0000.default.service.arpa
        Type: AAAA (IPv6 Address) (28)
        Class: IN (0x0001)
        Time to live: 10 (10 seconds)
        Data length: 16
        AAAA Address: fe80::dea6:32ff:fef3:954a
```

Thread Info:
PANID: 0x703D
Thread Network Key: A4D6078BDD1929443E9246CE7BC16C40
[Thread Sniffer log](https://github.com/openthread/openthread/files/11350386/google_thread_srp_AAAA_issue.zip)

Abhayakara, 2023-04-28:
Ah. So the bug is in the Google discovery proxy. It should not be returning link-local addresses from other links. It should be returning all AAAA records in the additional section though. Is it not?

fuxiaoming-lumi, 2023-05-04:
Yes. This is what I observed it.  It does not return all address information on the additional records. I am checking the thread spec and certification test plan, but I have not found the relevant requirements and test cases to ensure that the problem I encountered does not occur. 

In addition, I also found another issue on smartthings hub v3. It does not contain answer information in the __query response__. 
```
Frame 1262: 91 bytes on wire (728 bits), 91 bytes captured (728 bits) on interface \Device\NPF_Loopback, id 0
Null/Loopback
Internet Protocol Version 4, Src: 127.0.0.1, Dst: 127.0.0.1
User Datagram Protocol, Src Port: 57277, Dst Port: 49999
NXP 802.15.4 Sniffer Protocol
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xc000
6LoWPAN
Internet Protocol Version 6, Src: fd0b:7831:4c78:0:4d56:b953:cc06:92bd, Dst: fd0b:7831:4c78:0:a326:a042:e2a4:1eb1
User Datagram Protocol, Src Port: 53, Dst Port: 49154
Domain Name System (response)
    Transaction ID: 0x6236
    Flags: 0x8000 Standard query response, No error
    Questions: 2
    Answer RRs: 0
    Authority RRs: 0
    Additional RRs: 0
    Queries
        21C4E71E8CA1BE12-0000000000000001._matter._tcp.default.service.arpa: type SRV, class IN
        21C4E71E8CA1BE12-0000000000000001._matter._tcp.default.service.arpa: type TXT, class IN
    [Retransmitted response. Original response in: 1143]
    [Retransmission: True]
```
Maybe I missed some important information, but how can we ensure the __Interoperability__ at Thread. 
@jwhui @Abhayakara

fuxiaoming-lumi, 2023-05-04:
@Abhayakara Let me summarize all of the above. Here is my experience in different ecosystems.

__Apple(with homepod mini):__
When a Thread device queries for an SRV+TXT Record, AAAA record is not included in the query response. So it need to do the additional AAAA query. Commit [fa3563a](https://github.com/project-chip/connectedhomeip/commit/fa3563a9ea15f9baa34b3cb4434e4131f5e33643) is used to implement this mechanism.

__Google(with Nest Hub Gen2):__
When a Thread device queries for an SRV+TXT Record, AAAA record is included in the query response. But this AAAA record from query response is link local address. Please check the sniffer Log No.32127. However, if do the additional AAAA query, it will return the different AAAA record which is Unique Local Address. Please check the sniffer Log No.32161.
Thread Info:
PANID: 0x703D
Thread Network Key: A4D6078BDD1929443E9246CE7BC16C40
[Thread Sniffer log](https://github.com/openthread/openthread/files/11350386/google_thread_srp_AAAA_issue.zip)

__SmartThings(with SmartThings Hub v3):__
When a Thread device queries for an SRV+TXT Record, there is no SRV or TXT record in the query response. That is, there is no answer information in the the query response. Please check the sniffer log No.1262.
SmartThings:
Channel:14
panid: 0x65aa
networkkey:7f200b0a0f236afb5818b2e302172909
[smartthing_srp_issue.zip](https://github.com/project-chip/connectedhomeip/files/11394023/smartthing_srp_issue.zip)
 
__Alexa(with Echo Gen4):__
When a Thread device queries for an SRV+TXT Record, there is no SRV or TXT record in the query response. That is, there is no answer information in the the query response. Please check the sniffer log No.1714.
SmartThings:
Channel:11
PANID: 1713
NetworkKey:5392c9b34603a739554cda30592a4665
[alexa-srp-issue-1714.zip](https://github.com/project-chip/connectedhomeip/files/11396071/alexa-srp-issue-1714.zip)

Abhayakara, 2023-05-04:
Okay, in the last case this is because it's invalid to have two questions in a DNS lookup. The Google implementation and the Apple implementation both respond to both questions, but they aren't required to—the behavior in this case is undefined. So that's why the Echo and the SmartThings hubs are responding as they are. Although they should be returning an rcode of FormErr if they aren't going to handle the two-question query. As it is, this response seems to be saying that there is no TXT or SRV record under that name, but that the name exists (which I suppose is also possible).

The behavior of the Nest hub just looks like a bug. It would be wrong to add code to work around it. Similarly, it would be wrong to add code to work around the Echo and SmartThings failures. In all three cases, you have an answer, but it's the wrong answer, and that's why it doesn't work. The right thing to do is for Google, Nest and SmartThings to fix their implementations.

Whether that's the expedient thing I can't say. I am somewhat regretting implementing the two-question query response at this point—my goal was to avoid a train wreck on release, but it looks like we haven't done that.

fuxiaoming-lumi, 2023-05-05:
Thank you. Yes, I agree that we shouldn't add code to workaround that. I just wanted to see what would happen if I did additional AAAA records for this case. 

Matter1.0 supports Binding, Multi-fabric, OTA. 
1. Commission Matter Thread Accessory to Google/Apple/SmartThings/Alexa
2. Share this Accessory to our Vendor fabric
3. Want to trigger OTA or Binding in our Vendor fabric
4. At least for now, I did run into some issues and it is difficult to solve on the device side. However, commit [fa3563a](https://github.com/project-chip/connectedhomeip/commit/fa3563a9ea15f9baa34b3cb4434e4131f5e33643) do help my use case with Apple Ecosystem.

I believe these issues will also block the Intermittently Connected Devices (ICDs) feautre.

fuxiaoming-lumi, 2023-05-05:
I think we also need relevant test cases and certification requirements, especially for thread border routers, as there will be more and more thread border router products on the market.

jwhui, 2023-05-05:
> Okay, in the last case this is because it's invalid to have two questions in a DNS lookup. The Google implementation and the Apple implementation both respond to both questions, but they aren't [...]

OpenThread's current DNS Client implementation defaults to sending separate queries for SRV and TXT if the combined query fails. But it does currently depend on receiving an error from the server.

closed 2023-07-02 by fuxiaoming-lumi