project-chip/connectedhomeip #26292: [BUG] Android build pasesession fail when I use the pairing code from the open-commission-window

luobosu, 2023-04-28:
### Reproduction steps

1、I use App-A complete matter commission
2、Use App-A open the commission window
3、capture the qrcode from the App-A
4、Use App-B parse the qrcode , the get the device IP .    App-B use the chipDeviceController#discoverCommissionableNodes and chipDeviceController.getDiscoveredDevice(i) method to find the IP which discriminator is match.
5、get IP success
6、try  build PASE session with IP , but meet time-out issue.    Using chipDeviceController#pairDeviceWithAddress method.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

TAG 1.0.0.2

### Platform

android

### Anything else?

There is the logs:

2023-04-28 14:50:23.505 21468-21468/com.tplink.tpm5.trial D/NsdManagerServiceBrowser: Starting service discovering for '_matterc._udp'
2023-04-28 14:50:23.505 21468-10283/com.tplink.tpm5.trial D/DL: _OnPlatformEvent
2023-04-28 14:50:23.512 21468-10282/com.tplink.tpm5.trial I/NsdManagerServiceBrowser: Started service '_matterc._udp'
2023-04-28 14:50:23.513 21468-10282/com.tplink.tpm5.trial I/NsdManagerServiceBrowser: Found service 'FF106CFB6207D4BC'
2023-04-28 14:50:23.513 21468-10282/com.tplink.tpm5.trial I/NsdManagerServiceBrowser: Found service '46986CE91C6F94F1'
2023-04-28 14:50:28.509 21468-10282/com.tplink.tpm5.trial W/NsdManagerServiceBrowser: Succeed to stop discovery service '_matterc._udp
2023-04-28 14:50:28.509 21468-21468/com.tplink.tpm5.trial D/NsdManagerServiceResolver: resolve: Starting service resolution for 'FF106CFB6207D4BC' type '_matterc._udp'
2023-04-28 14:50:28.510 21468-21468/com.tplink.tpm5.trial D/NsdManagerServiceResolver: resolve: Starting service resolution for '46986CE91C6F94F1' type '_matterc._udp'
2023-04-28 14:50:28.513 21468-10282/com.tplink.tpm5.trial W/NsdManagerServiceResolver: Failed to resolve service '46986CE91C6F94F1': 3
2023-04-28 14:50:28.517 21468-10282/com.tplink.tpm5.trial I/NsdManagerServiceResolver: Resolved service 'FF106CFB6207D4BC' to /fe80::32de:4bff:fef6:8ed
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [0] : D 1682
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [1] : T 1
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [2] : CM 2
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [3] : DN Mini Smart Wi-Fi Plug
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [4] : DT 266
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [5] : PH 36
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [6] : PI NULL
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [7] : RI 0200C24B04AC974EA68427082E5701114313
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [8] : VP 4488+257
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [9] : SAI 300
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS:  ----- entry [10] : SII 5000
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: Discovered node:
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Hostname: fe80::32de:4bff:
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	IP Address #1: fe80::32de:4bff:fef6:8ed
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Port: 5540
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Mrp Interval idle: 5000 ms
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Mrp Interval active: 300 ms
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	TCP Supported: 1
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Rotating ID: 0200C24B04AC974EA68427082E5701114313
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Device Name: Mini Smart Wi-Fi Plug
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Vendor ID: 4488
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Product ID: 257
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Device Type: 266
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Long Discriminator: 1682
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Pairing Hint: 36
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Instance Name: FF106CFB6207D4BC
2023-04-28 14:50:28.518 21468-10282/com.tplink.tpm5.trial D/DIS: 	Commissioning Mode: 2
... ...
2023-04-28 14:51:13.610 21468-9475/com.tplink.tpm5.trial D/CTL: newDeviceController() called
2023-04-28 14:51:13.613 21468-9475/com.tplink.tpm5.trial D/CTL: Stopping commissioning discovery over DNS-SD
2023-04-28 14:51:13.613 21468-9475/com.tplink.tpm5.trial D/CTL: Setting attestation nonce to random value
2023-04-28 14:51:13.613 21468-9475/com.tplink.tpm5.trial D/CTL: Setting CSR nonce to random value
2023-04-28 14:51:13.613 21468-9475/com.tplink.tpm5.trial D/CTL: Setting attestation nonce to random value
2023-04-28 14:51:13.613 21468-9475/com.tplink.tpm5.trial D/CTL: Setting CSR nonce to random value
2023-04-28 14:51:13.613 21468-9475/com.tplink.tpm5.trial D/CTL: add paa from paaArray:38
[... 89 lines cut ...]
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: pairDeviceWithAddress() called
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/IN: SecureSession[0xb400007b293d9de0]: Allocated Type:1 LSID:32358
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/SC: Assigned local session key ID 32358
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/EM: <<< [E:42452i M:240388959] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/IN: (U) Sending msg 240388959 to IP address 'UDP:[fe80::32de:4bff:fef6:8ed]:5540'
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/SC: Sent PBKDF param request
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting failsafe timer from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting adminSubject from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting attempt thread scan from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting attempt wifi scan from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting attestation nonce from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting CSR nonce from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Setting PASE-only commissioning from parameters
2023-04-28 14:51:13.731 21468-9475/com.tplink.tpm5.trial D/CTL: Commission called for node ID 0x0000000000000025
2023-04-28 14:51:14.108 21468-10283/com.tplink.tpm5.trial D/EM: Retransmitting MessageCounter:240388959 on exchange 42452i Send Cnt 1
2023-04-28 14:51:14.108 21468-10283/com.tplink.tpm5.trial D/IN: (U) Sending msg 240388959 to IP address 'UDP:[fe80::32de:4bff:fef6:8ed]:5540'
2023-04-28 14:51:14.509 21468-10283/com.tplink.tpm5.trial D/EM: Retransmitting MessageCounter:240388959 on exchange 42452i Send Cnt 2
2023-04-28 14:51:14.509 21468-10283/com.tplink.tpm5.trial D/IN: (U) Sending msg 240388959 to IP address 'UDP:[fe80::32de:4bff:fef6:8ed]:5540'
2023-04-28 14:51:15.145 21468-10283/com.tplink.tpm5.trial D/EM: Retransmitting MessageCounter:240388959 on exchange 42452i Send Cnt 3
2023-04-28 14:51:15.145 21468-10283/com.tplink.tpm5.trial D/IN: (U) Sending msg 240388959 to IP address 'UDP:[fe80::32de:4bff:fef6:8ed]:5540'
2023-04-28 14:51:16.140 21468-10283/com.tplink.tpm5.trial D/EM: Retransmitting MessageCounter:240388959 on exchange 42452i Send Cnt 4
2023-04-28 14:51:16.140 21468-10283/com.tplink.tpm5.trial D/IN: (U) Sending msg 240388959 to IP address 'UDP:[fe80::32de:4bff:fef6:8ed]:5540'
2023-04-28 14:51:17.647 21468-10283/com.tplink.tpm5.trial E/EM: Failed to Send CHIP MessageCounter:240388959 on exchange 42452i sendCount: 4 max retries: 4
2023-04-28 14:51:19.958 21468-10283/com.tplink.tpm5.trial E/SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
2023-04-28 14:51:19.959 21468-10283/com.tplink.tpm5.trial D/IN: SecureSession[0xb400007b293d9de0]: Released - Type:1 LSID:32358

bzbarsky-apple, 2023-04-28:
@luobosu Do you have the device log by any chance?  Did the device receive the PBKDFParamRequest message and fail to respond?  Or did it not receive it at all?

Was the discovered IP the right one?

luobosu, 2023-05-04:
Do you have the device log by any chance? Did the device receive the PBKDFParamRequest message and fail to respond? Or did it not receive it at all?
--- - - - - - -
Sorry, I can not fetch the device logs , so I can not judge whether the device receive PBKDFParamsRequest.

I think the IP is right ( fe80::32de:4bff:fef6:8ed) . It was discovered by matter sdk .
@bzbarsky-apple

bzbarsky-apple, 2023-05-04:
> Sorry, I can not fetch the device logs , so I can not judge whether the device receive PBKDFParamsRequest.

Well, then it's hard to say whether the problem is with the network or with the device.

luobosu, 2023-05-04:
> > Sorry, I can not fetch the device logs , so I can not judge whether the device receive PBKDFParamsRequest.
> 
> Well, then it's hard to say whether the problem is with the network or with the device.

this device can be connected by iOS matter sdk  ,  so I think the device is no error . 

Maybe it is Android SDK  issue or network issue ?

luobosu, 2023-05-10:
I found the issue.
I use Controller-A to call chipDeviceController#discoverCommissionableNodes  ,  and use  Controller-B to call chipDeviceController#pairDeviceWithAddress. It is wrong.

Before Controller-B to call chipDeviceController#pairDeviceWithAddress , it must call chipDeviceController#discoverCommissionableNodes , too.

closed 2023-05-10 by luobosu