project-chip/connectedhomeip #26294: [Platform][ASR] add asr582x & asr595x support

tx2rx, 2023-04-28:
add asr platform support

brief:
ASR582X is a low-power, high-performance, highly integrated combo SoC supporting 1x1 IEEE 802.11 b/g/n and BLE 5.1. The chip integrates RF transceiver, 802.11/BLE PHY + MAC, Arm China STAR-MC1 Processor, multiple peripheral interfaces, AOA/AOD, Real Time Counter (RTC) and a complete power management module.

ASR595X is a low-power, high-performance, highly integrated combo SoC supporting 1x1 IEEE 802.11 b/g/n/ax and BLE 5.1. Thanks to the uniquely designed power management circuitry, this chip significantly reduces power consumption and peripheral components.

tx2rx, 2023-04-28:
https://github.com/project-chip/connectedhomeip/pull/26295

mideayanghui, 2023-04-28:
Nice, keep go on

tx2rx, 2023-05-04:
https://github.com/project-chip/connectedhomeip/pull/26361

PR #26361 (2023-05-16): [ASR] add ASR582X & ASR595X support
Change overview

fix #26294 

- support asr582x
- support asr595x
- add asr platform src files
- add asr example platform files
- add third_party/asr/

- support lighting-app , other examples will be added later

- update build_examples.py support

Testing

How was this tested? (at least one bullet point required)

- new platform, Commissioning test pass by chip-tool

Products implemented through the ASR platform were certified.
closed 2023-05-16 by andy31415